supermerill/SuperSlicer #2077: first layer inlay

evils, 2021-12-17:
### What happened?

i'm trying to do an different colour inlay on the first layer
(mostly as described [here](https://www.reprap.org/wiki/Mulitcolour_printing_with_one_hotend))

several issues ensued
1. the layer preview seems to show both filaments as the same colour
1.1. this does not occur in prusaslicer 2.3.3
1.2. it momentarily rendered right before finalizing the preview
2. "The Wipe tower is only supported if all objects have the same variable layer height"
2.1. i only intend to change filament on the first layer, which is to say: all layers that make use of the wipe tower are the same height
2.2. i cannot set variable layer height across multiple objects
2.2.1. surely both objects could be considered as 1 larger object for the purposes of setting adaptive layer heights?
2.3. suggestion: allow having a wipe tower per object(s) if unmatched layer heights are needed

### Project file & How to reproduce

[inlay-test.3mf.zip](https://github.com/supermerill/SuperSlicer/files/7734796/inlay-test.3mf.zip)

### Version

2.3.57.6

### Operating system

NixOS-unstable

### Printer model

Prusa i3 MK3S+

supermerill, 2021-12-17:
Your tool-change gcode is "m600" and the slicer can't know it has changed to what filament with that.
Prusaslicer add a hardcoded `T{next_extruder}` but in superslicer, if you define your own toolchange gcode, it won't.
You can add the `T{next_extruder}` before the M600 in the toolchange gcode to have the good ol'dbehaviour.

evils, 2021-12-17:
thanks, that definitely fixed the display issue (though it's still odd it briefly displayed the expected view before showing the correct one)
(i'm not sure in what situations it makes sense not to change the tool to the tool changed to on a tool change though...)
(maybe if a filament change wasn't a tool change, but then it'd have to adjust the colour anyway)

2 notes in case someone ends up here looking to de first layer inlay
#818 is annoying so this may be the way forward (i'm assuming that'll work right with the first layer)
```
T{next_extruder}
{if layer_num > -1}M600{endif}
```
the order of operations depends on the order of the configured extruders (so items set to print with extruder 1 print first, then extruder 2)
(i expected it to do them in the order they're in the file hierarchy, but i suppose this is a more foolproof way to reduce the number of required tool changes)

first test worked quite well
[image]
)

supermerill, 2021-12-17:
> it's still odd it briefly displayed the expected view before showing the correct one

Its probably the slicing preview. It's a preview based on the sliced information (not gcode). You can get it view the tabbed layout (not compatible with many os) or by modifying some printer or filament setting (like a custom gocde or filament temp, or a name)  after a successful slicing.

evils, 2022-02-13:
closing this as i've found it's possible to group items and do variable layer height for the group (this solves point 2)

and it was pointed out to me i don't need a wipe tower for this
(a second purge line with lifts before and after may be a good idea though)

a remaining sore point is in how it deals with small text, but that's hardly specific to my use
[image]
(D-DIN font shown)

prusaslicer without its thin wall detection is far worse
[image]

prusaslicer with thin wall detection arguably yields the best result
[image]
but they currently have a bug (7821) that almost caused a machine crash, so that's not an option for me

closed 2022-02-13 by evils

supermerill, 2022-02-13:
for the extremly thin thin walls, it's a bug in the "merge thin wall with perimeter" that is fixed in the nightly

evils, 2022-02-13:
here's a zip with 3 STLs
one with walls significantly thinner than is actually printable (Relief SingleLine font)
one with lines ~0.4mm (SingleLine padded to printable width)
and one with D-DIN font
each using a bunch of characters that seem to be most failure prone in an array of sizes
[samples.zip](https://github.com/supermerill/SuperSlicer/files/8055767/samples.zip)

superslicer sometimes decides these line widths suffice to print 2 lines / a perimeter, while my goal is to get all text as 1 line without missing pieces

(side note, if there was a way to more directly couple the [Relief SingleLine](https://github.com/isdat-type/Relief-SingleLine) font to the slicer, that would be awesome (it's a font specifically made for applications like milling and laser cutting, where there is a single line being drawn))

evils, 2022-02-14:
and since you asked for a project file, here's those STLs in a 3mf
[3mf-samples.zip](https://github.com/supermerill/SuperSlicer/files/8057432/3mf-samples.zip)