supermerill/SuperSlicer #2078: Filament Temperatur Calibration is missing temp changes in macOS ARM version

sofacoder, 2021-12-17:
### Version
2.3.57.7

### Operating system type + version
macOS Monterey 12.0.1 (Apple MBP M1 Pro)

### Behavior

The generated G-code for the filament temperature calibration misse the M104 commands in the ARM version. It is in the G-Code when I use the Intel-version on the same system (running under rosetta)

Reproduction:
- generate the G-Code for the filament temperature calibration 
- look at the G-Code for the M104 commands 
- with the ARM version there are missing for me [see this file](https://github.com/supermerill/SuperSlicer/files/7737390/Temperature.calibration.gcode.zip)
- with the Intel version those code are there. [See this file](https://github.com/supermerill/SuperSlicer/files/7737398/Temperature.calibration-2.gcode.zip)

_Is this a new feature request?_
Related guides for writing feature requests: http://meta.stackexchange.com/a/259196 http://nickohrn.com/2013/09/write-great-feature-request-bug-report/

#### Project File (.3MF) where problem occurs
The project is generated with Super Slicer itself, but here are the different results that I would expect to be the same:
- ARM: [Temperature calibration.gcode.zip](https://github.com/supermerill/SuperSlicer/files/7737390/Temperature.calibration.gcode.zip)
- Intel: [Temperature calibration-2.gcode.zip](https://github.com/supermerill/SuperSlicer/files/7737398/Temperature.calibration-2.gcode.zip)

supermerill, 2021-12-19:
mmm
That's unexpected.
It's really not here or it's a display/gui not refreshing problem?

sofacoder, 2021-12-19:
We’ll see the G-code files. I did the same in both apps. Generated the calibration assets, let in live and export the G-code.

supermerill, 2021-12-19:
Oh, yeah.
They are not here.... really wondering why.

supermerill, 2021-12-19:
If you save the calibration project from the intel version and open it into the arm one, does the gcode contains `M104 S230 ; floor 1 of the temp tower set` ?