supermerill/SuperSlicer #2082: Show Purge Line (G-Code start) on preview

marc-gist, 2021-12-19:
**Is your feature request related to a problem? Please describe.**

It would be great to have the purge line display in the sliced preview so that on large/odd shaped objects we can make sure it isn't going to interfere with the resulting print.  I.e. the skirt and/or object may go through the purge line, would be really bad.

**Describe the solution you'd like**

On the preview of the g-code, show the purge line steps/placement.

**Describe how it would work**

Hopefully self explanatory?  display where the purged filament will be in the g-code preview.

**Describe alternatives you've considered**
picture in my head.  however, I'm likely to forget, as most of the time it isn't an issue.  So I get this is likely low priority.

**Additional context**
S3D (shudder, I know), shows the purge line :)

gpread, 2021-12-19:
PrusaSlicer 2.4.0 now does this, so SuperSlicer should have it also as soon as its merged

PedroC1999, 2021-12-19:
To note: If you use Klipper it can be common to have the purge line relocated to a macro in the firmware, and this would not show up in the slicer