supermerill/SuperSlicer #2085: Feature to get rid of tangent lines next to holes

laptop-laptop, 2021-12-19:
**Is your feature request related to a problem? Please describe.**

I have been having issue with the kind of tangent lines mentioned in the below thread for over a year. Even with ironing and monotonic infill for top, bottom and solid under 2.3.57.7+win64, they are still there. Are they unavoidable due to the way Slic3r works? Is there any good way to get rid of them or make them less visible?

https://forum.prusaprinters.org/forum/original-prusa-i3-mk3s-mk3-how-do-i-print-this-printing-help/anyone-know-what-could-cause-these-stray-tangent-lines/

Will adding another ironing patterns improve the situation?

supermerill, 2021-12-19:
Did you try the superslicer ironing pattern?

laptop-laptop, 2021-12-20:
Where is it?

supermerill, 2021-12-20:
infill -> top pattern -> ironing (in the combo box)