supermerill/SuperSlicer #2086: Tool Change G-code isn't parsing some G0 commands correctly

woodencase01, 2021-12-19:
### What happened?

I'm trying to use an IDEX setup and have the hotends prime and wipe during the tool change.

I've used this script:
```; Tool change
{if use_relative_e_distances}G0 F180 E-2;{else}M83;
G0 F180 E-2; Retract
M82;{endif}
{if previous_extruder==0}G0 F6000 X-15; Park{else}G0 F6000 X237; Park{endif}
T[next_extruder]; Switch extruder
{if next_extruder==0}G0 F6000 X-4; Prime zone 11mm from limit switch{else}G0 F6000 X224; Prime zone 13mm from limit switch{endif}
{if use_relative_e_distances}G0 F180 E4; Prime{else}M83; Prime
G0 F180 E4;
M82;{endif}
{if next_extruder==0}G0 F6000 X-8; Wipe{else}G0 F6000 X228; Wipe{endif}```

The problem starts here:
```{if next_extruder==0}G0 F6000 X-4; Prime zone 11mm from limit switch{else}G0 F6000 X224; Prime zone 13mm from limit switch{endif}```

As an example, here is one output I get in the Gcode:
```T1; Switch extruder
 X224; Prime zone 13mm from limit switch```

And should be:
```T1; Switch extruder
 G0 F6000 X224; Prime zone 13mm from limit switch```

Same problem happens with T0.
It seems like "G0 F6000" is getting ignored. 
I tried many workaround and it didn't solved my problem.

Copying the "G0 F6000 X224" in comment works:
```{if next_extruder==0}; G0 F6000 X-4
G0 F6000 X-4; Prime zone 11mm from limit switch{else}; G0 F6000 X224
G0 F6000 X224; Prime zone 13mm from limit switch{endif}```

I get:
```T1; Switch extruder
; G0 F6000 X224
 X224; Prime zone 13mm from limit switch```

### Project file & How to reproduce

[Igus Bushings.zip](https://github.com/supermerill/SuperSlicer/files/7741267/Igus.Bushings.zip)

### Version

Version 2.3.57

### Operating system

Win10

### Printer model

Climber 7

supermerill, 2021-12-19:
project file?

woodencase01, 2021-12-20:
I had to zip it in this file, I can't load 3mf files:
[Igus Bushings.zip](https://github.com/supermerill/SuperSlicer/files/7741267/Igus.Bushings.zip)

I think the code block has overriden the hyperlink on my first message.

I found patch to make it work:
```{if next_extruder==0}G0 F6000 G0 F6000 X-4; Prime zone 11mm from limit switch{else}G0 F6000 G0 F6000 X224; Prime zone 13mm from limit switch{endif}```
And the output is the ok.

supermerill, 2021-12-20:
put the `F` after the X or Y in a G1/G0 command, or it will be culled by the "reduce gcode size" cleaning pass. It assumes that if there is a F, it's the last thing useful on the line.

woodencase01, 2021-12-20:
Oh, I didn't know.

It works as intended now!

Thanks for your help!

closed 2021-12-20 by woodencase01