supermerill/SuperSlicer #2087: [FR] Retract Before Layer Change

CCS86, 2021-12-19:
To me it makes sense to invert the order of the retract and layer change moves:

[image]

My reasoning is that because the layer change move acts to "open" the nozzle and give it a place to ooze, the retraction move should occur first to relieve nozzle pressure.

CCS86, 2021-12-19:
I noticed it more in this case because I had left a bit of custom layer change gcode in place (M593 P"none"), which causes a slight delay.

supermerill, 2021-12-20:
currently, you have to check "retract on layer change" to do that.

CCS86, 2021-12-21:
Am I correct that enabling this option forces a retract, even if the last layer ending point is very close to the next layer starting point, so no significant travel is needed?

Would you be willing to invert this order of operations in a future release?

supermerill, 2021-12-21:
Yes
Yes

edgauthier, 2022-12-16:
I'd love to see this (swapping the order of the retraction and layer change). I generally have 'retract on layer change' disabled as it seems to cause excessive retractions. Swapping the order as @CCS86 suggested seems like a good compromise from a laymans perspective.