supermerill/SuperSlicer #2088: Default extrusion width per model modifiers do not work, unless extrusion width is set to zero

Angelol125, 2021-12-19:
### What happened?

I am currently testing extrusion widths and when I tried to test 2 different ones, 0.4 and 0.45 mm since I'm using a 0.4 nozzle, I went ahead and created two 20mm cubes from the add a shape menu. I tried adding a default extrusion width modifier to both of them, but that didn't work as expected at all, except only under one unique occasion. I tried setting the modifiers both in mm and percentage modes, namely 0.4/0.45 and 100%/120% respectively, but that didn't work at all. After that, I tried changing my extrusion width on my Print Settings, from 100% to 0.4 mm, and again tried the percentage/mm values in the modifiers to no avail. Finally, only after setting my extrusion width to 0 in my Print settings, the results were as expected.

### Project file & How to reproduce

attached here is the project file

[Extrusion Width Test.zip](https://github.com/supermerill/SuperSlicer/files/7741619/Extrusion.Width.Test.zip)

### Version

2.3.57.7

### Operating system

Windows 10 64 bit

supermerill, 2021-12-19:
Default extrusion width is only used to replace another width field that is set to 0 and if the default is set to something different than 0.

% is % of nozzle diameter.

You have to modify each width you want to modify individually in the modifier.

Angelol125, 2021-12-19:
I'm sorry, I don't think I understand. These are my print settings:

[image]

When I try to add an extrusion width modifier for a model, while having a default extrusion width as shown in my settings above, the modifier doesn't work, even if I try 120% or 0.48mm like so:

[image]

supermerill, 2021-12-20:
did you read the tooltips, all of them?

neophyl, 2021-12-20:
When they are set to 0 then they are based off the default, so it does have a cascade effect.  So setting to 0 then modifying the default extrusion width results in a change to the rest.

However when set to specific percentages the 'base' value is no longer default extrusion width but is nozzle diameter.  The tooltip does state this.   As such changing default extrusion width with a modifier does NOT trickle down to the other extrusion types and so you don't see any change.
If you load up modifiers for Perimeter Extrusion width and set that to a % value, for example 120% then its width will be 0.4x120% (assuming a 0.4mm nozzle).  While the rest of the extrusion types will still be 0.4x100%.  When you set them all with extra modifier entries then they change.  

In effect 0 is a special case.

supermerill, 2021-12-20:
> I honestly think the wording in the description/tooltips could be better. 

Yes, I can change it if someone proposes better wording.

> I only wanted to modify solid_infill_extrusion_width.

Sorry, I didn't understand that at all when reading your first post.

> this new modifier would only work if the default_extrusion_width of the model is also set to 0

what? no... I really understand less and less.

Angelol125, 2021-12-20:
> > I honestly think the wording in the description/tooltips could be better.
> 
> Yes, I can change it if someone proposes better wording.
> 
> > I think using math formulas and conditions instead of words when trying to describe this behavior in a tooltip, results in an easier understanding of how things are supposed to work.
> 
> I can copy/paste the source code. If you need it to understand.
>

The formulas I wrote in my previous reply are a better description imho, that's why I suggested using them as a tooltip. 

> > I only wanted to modify solid_infill_extrusion_width.
>
> Sorry, I didn't understand that at all when reading your first post.
> 
> > this new modifier would only work if the default_extrusion_width of the model is also set to 0
> 
> what? no... I really understand less and less.

  - default_extrusion_width=100% 
       OR 
  - default_extrusion_width=0 

Both 100% and 0 when combined with a 0.4 nozzle_diameter, will result in default_extrusion_width=0.4. The problem is that if I add a solid_infill_extrusion_width modifier, that will only work if default_extrusion_width=0, not with 100%. That's why I suggest creating a prompt to either add a model specific (local) default_extrusion_width (by adding a new modifier) with a value of 0, or change the tab to Print Settings and indicate the default_extrusion_width textbox with an arrow, so the user can change the global default_extrusion_width to 0.

neophyl, 2021-12-20:
I’m not sure I’m following either.  If you want solid infill extrusion width to be say 120% then add a modifier for that. That will work even if it’s set to 100% in the normal place. The reason it wasn’t working for you in your example is that your modifier was for the default extrusion width, which as the others aren’t based on that when set to % means they didn’t change. 

I tried it earlier with 2.3.57.7 and with all the extrusion widths set to 100% in print settings I could add modifiers and change any of them that have modifiers to use a different %. You do know there are extra modifiers available that aren’t in the “common” lists ?

supermerill, 2021-12-20:
also, try to forgot the default_extrusion_width. I didn't removed it to avoid breaking some configuration, but it's almost useless, don't use it.

Angelol125, 2021-12-20:
I'm sorry, I must have missed the reply notifications.

It seems that if someone wants to change the extrusion width for all feature types with a single modifier, to make them 0.5 for example, he must then use a default_extrusion_width=125% modifier while having all the other global extrusion widths in Print Settings tab set to 0. What confuses me is that default_extrusion_width modifier has no impact if all the other extrusion widths are set to 100% in Print Settings tab. If they are set to anything but 0, the modified default_extrusion_width does not work.

neophyl, 2021-12-20:
That's correct.  I think only time the other extrusion widths are actually based on the default extrusion width is when they are set to zero.  As SuperMerill points out its pretty much a deprecated value that is only kept around to not break some parts of the code.  Along with that zero special case.

If you want to change the extrusion widths when not set to zero then you have to add multiple entries into the modifier.
However given that its easy to copy modifiers from one modifier to another that's not as bad as it could be.

Angelol125, 2021-12-20:
Again thanks for the help @neophyl and really sorry for creating an issue where there wasn't one @supermerill. Feel free to close the issue as it seems there is no reason for one in the first place.

supermerill, 2021-12-21:
I've updated the gui to disable the field when it's not used by other settings.