supermerill/SuperSlicer #2090: avoid_crossing_perimeters crosses through other materials

Juusoz, 2021-12-20:
### What happened?

So, a little problem, avoid_crossing_perimeters crosses through other materials, leaving streaks across your print.
Probably caused by the code treating the other material as part of the same print instead of no-go areas.

[image]

[image]

### Project file & How to reproduce

It should do it on any print that has a second material on the same layer that the nozzle can cross through.

**Project file:**
[streaks.zip](https://github.com/supermerill/SuperSlicer/files/7742024/streaks.zip)

### Version

2.3.57.7

### Operating system

Windows 10

### Printer model

Biqu B1

supermerill, 2021-12-20:
Didn't think of that.
Will fix it later... much later.

Juusoz, 2022-01-30:
> Didn't think of that. Will fix it later... much later. use lift in the meantime.

I've been trying to tweak this same print for a while now, and it seems that lift doesn't actually help, because it doesn't actually always lift when there is another material next to it.

[image]
[image]

It doesn't seem to do it when the yellow area is printed by itself though, only when there is another print surface next to it.

[image]

supermerill, 2022-01-30:
lift only trigger on retract, so be careful to not activate options that remove rectration.