supermerill/SuperSlicer #2091: macOS ARM binary broken

taglia, 2021-12-20:
### What happened?

The macOS ARM binary does not work on a new MacBook Pro (with the M1 Max CPU), the following error appears when trying to launch it:

`“SuperSlicer” is damaged and can’t be opened. You should eject the disk image.`

[image]

The Intel binary works perfectly with Rosetta 2.

### Project file & How to reproduce

Download the macOS binary from the GitHub release page, open the DMG file, double click on the SuperSlicer app package.

### Version

2.3.57.7

### Operating system

macOS 12.1 (21C52)

supermerill, 2021-12-20:
duplicate of #2055

taglia, 2021-12-21:
My bad, apologies for not searching properly, and thanks for pointing me to the solution.

closed 2021-12-21 by taglia