supermerill/SuperSlicer #2101: Modifying a printer profile makes slicer to not produce the  right gcode

Tinchus2009, 2021-12-21:
### Version
2.3.56.9

### Operating system type + version
WIN11

### Behavior
I have several prfiles for several material. By accident I erased the printer profile. I created a new one, I made sure the profile and material dependencies were pointing at that profile. I sliced and then when gcode was exported, all temperatures were set to 200 degrees, like if the defaul profile was being used and not the chosen one (I made sure I choose the profile in all tabs, also the material was selected, Im sure of that)

I imported a config bundle back up and the error was fixed. But any time I modified the printer profile this happens. Is like a default ghost material is being chose at exporting time.

Any ideas?

supermerill, 2021-12-21:
A project file where this happen?

Tinchus2009, 2021-12-22:
Sorry for that. I have attached a sample project file. Exporting that gcode gives me the gcode also attached. Note that in the gcode all temperatures are 200 degrees, like if the material is not being used
[Paint_Mixer.zip](https://github.com/supermerill/SuperSlicer/files/7763346/Paint_Mixer.zip)
?

supermerill, 2021-12-22:
In this project , the filament profiles all have a temperature of 200, that's why it print a 200°.

I think I need more context to understand the real problem.
You can't select your other filaments?

Tinchus2009, 2021-12-22:
that is the problem: I dont have a filament with 200 degrees. And the selected filament in that project is using 420 degrees.
I can select any of my filament, but no matter wich one I select, the gcode is always using 200 degrees.

supermerill, 2021-12-22:
Can you zip your configuration directory?
I think it's needed to replicate this problem.

Tinchus2009, 2021-12-22:
[config.zip](https://github.com/supermerill/SuperSlicer/files/7764918/config.zip)

Tinchus2009, 2021-12-22:
Note. I have updated to 2.3.57.7 version and Im having the same problem. I erase the prusa printer profile, wich I dont have, and fro there any of my profiles with any of my materials, the generated gcode just set s 200 degrees temp....

supermerill, 2021-12-23:
Everything seems normal and expect for me. No 200/200 temp.
When i change the print, the filament change as well and its temperature is good.

If I import your project file, it is 200/200, but it's saved this way and reference the -default- filament.

I don't know how to go forward...
maybe you can tell me the steps to reproduce it. Or a video.

Tinchus2009, 2021-12-23:
well. I dont know hot to proceed either. It is not wrking for me. Lets forget about the version I was using, I have downloaded latest version, I imported the config bundle to have my profiles, and it is doing the same thing.

Here is what I do:
I import and STL, I select the profile, the material, I export the gcode... and the temperatures used in the gcode are 200 degrees. No material in my material list uses 200 degrees.

Here are screenshots so you ca see each step:

[image]

Material chosen:

[image]

Printer chosen:

[image]

Slice done:

[image]

I have attached the project file and the generated gcode for this:

[Shape-Box.zip](https://github.com/supermerill/SuperSlicer/files/7769865/Shape-Box.zip)
[Shape-Box.zip](https://github.com/supermerill/SuperSlicer/files/7769870/Shape-Box.zip)

supermerill, 2021-12-23:
I suceed to reproduce it! I'll fix it.
In the meantime, it seems to work if you uncheck `suppress 'default' presets` in preferences