supermerill/SuperSlicer #2102: Gap fill causes Gcode output ;WIDTH:-nan(ind) G1 F-nan(ind)

Thegribbs, 2021-12-21:
### What happened?

Enabling gap fill on a specific print causes the Gcode to output a nan at certain sections of code.
I've not had this with other prints, and I'm not sure what settings are causing this on this specific project, or why they've suddenly appeared.
Turning off gap fill fixes the issue.

Here is an example-

`G1 X195.651 Y187.395
G1 X195.254 Y187.243
G1 X194.989 Y187.172
G1 X184.342 Y187.102
G1 Z7.56
G1 E0.5 F4800
;WIDTH:0.359299
G1 F900
;WIDTH:-nan(ind)
G1 F-nan(ind)
G1 X184.189 Y187.158 F10800
;WIDTH:0.221179
G1 F900
G1 X184.342 Y187.102 E0.002
G1 X184.189 Y187.047 E0.002
;WIPE_START
G1 F8640
G1 X184.342 Y187.102 E-0.08608
G1 X184.189 Y187.158 E-0.08608
;WIPE_END`

### Project file & How to reproduce

[r2.zip](https://github.com/supermerill/SuperSlicer/files/7758808/r2.zip)

### Version

2.3.57

### Operating system

Windows 10

### Printer model

Ratrig V-Core 3 (klipper)

supermerill, 2021-12-22:
not obvious, but a duplicate of #2099

macchinette, 2021-12-23:
I also wanted to report the problem with a specific project and specific settings.
I confirm that disabling the gap_fill the problem does not arise.
This also happens by keeping the gap_fill active, but by varying the layer height with a value to be found with various tests.
For this project at 0.18 the problem disappears.

Thank you

Windows 10
Version: 2.3.58.0

[image]
[199060.zip](https://github.com/supermerill/SuperSlicer/files/7769277/199060.zip)

kesor, 2021-12-27:
Had the same happen to me today with the latest version

```
;TYPE:Gap fill
;WIDTH:-nan
G1 F1800
G1 X204.359 Y141.738 E-nan
G1 X201.747 Y139.126 E-nan
G1 X201.718 Y139.095 E-nan
G1 E-0.5
G1 Z13.825 F27000
G1 X214.677 Y160.903
G1 Z13.625
G1 E0.5 F1800
```
I uploaded the full sliced gcode file here https://gist.github.com/kesor/6b8c2073778e33055c60c8da14961605

supermerill, 2022-01-04:
@kesor please test with the nightly. Or at least put the project file. The gcode file isn't useful in this case.

macchinette, 2022-01-04:
> @kesor si prega di testare con il serale. O almeno mettere il file di progetto. Il file gcode non è utile in questo caso.

Hi Merill, in my project file with the night version the problem is solved.
Thanks so much

closed 2022-04-01 by supermerill