supermerill/SuperSlicer #2103: Anycubic Linear Plus not able to add Flex(TPU) to filament pull down.

brosskgm, 2021-12-22:
Anycubic Linear Plus not able to add Flex(TPU) to filament pull down.

Ver: 2.3.57.7

Tried to add filament with Plus selected but did not show Flex available.
Tried selecting (All) but will not add Flex to Linear Plus pull down filament choices.

Thanks
Bob

supermerill, 2021-12-22:
in configuration-> preferences, select "Show incompatible print & filaments presets" (in presets category)
then selct your filament in the dropdown.
in the filament setting, go to Dependencies section
check if it isn't enforce only on some printer (unlikely) or if it has some rules in "compatible XXX condition" that prevent it for showing up with your printer (likely)

brosskgm, 2021-12-22:
Thanks. I'll give this a try.

brosskgm, 2021-12-22:
Still not giveing Flex under Linear Plus in the filament profile section.

closed 2021-12-22 by brosskgm

supermerill, 2021-12-22:
you closed the issue, so everything is good? Because it doesn't seems like it in your last message...

brosskgm, 2021-12-22:
Didn't want to leave it open to long. I only just started looking at the slicer and it could be a while between trys and didn't want to do that to anyone.

supermerill, 2021-12-22:
If you talk about the wizard and not the main GUI, then that means that nobody has created a profile for a tpu filament for your printer.
You can create one yourself.
If you want o get the one created for the prusa mk3 (for example), you have to add the prusamk3 to be able to install these ones.

brosskgm, 2021-12-22:
Ok thanks. I'll work on that.