supermerill/SuperSlicer #2104: Bug: Tool path outside print area in latest SuperSlicer version for ARM

Dreistein75, 2021-12-22:
### What happened?

I enclose the project file here. This part cannot be sliced because there is an wrongful claim that there is a tool path detected outside of the print area (which is DEFINITELY not the case). The same project opened in PrusaSlicer works just fine.
This bug is in Version 2.3.57.7 for ARM (Apple M1 chip).

### Project file & How to reproduce

[SuperSlicer_OutsideError.3mf.zip](https://github.com/supermerill/SuperSlicer/files/7759288/SuperSlicer_OutsideError.3mf.zip)

### Version

2.3.57.7

### Operating system

MacOS (on M1 chip)

### Printer model

Voron 2.4 with AB

Dreistein75, 2021-12-22:
Edit: I just downloaded the same SuperSlicer version for non-ARM MacOS and the project slices perfectly also, i.e. it's another ARM issue (sadly)

supermerill, 2021-12-22:
Yes, probably something creating an unexpected NaN or Inf because of arm...
Can't debug it without an Arm hardware and a good debugger.

Troncooooo, 2022-04-30:
I've the same issue on linux with SuperSlicer-ubuntu_18.04-2.3.57.12.AppImag. The previous version SuperSlicer-ubuntu_18.04-2.3.57.11.AppImage doesn't give the error.
[buisklem_v1.obj.zip](https://github.com/supermerill/SuperSlicer/files/8596666/buisklem_v1.obj.zip)