supermerill/SuperSlicer #2105: Add parameter display on the cursor when hovering over an sliced model

DrDaveone, 2021-12-22:
The display of parameters on the mouse cursor is very lacking when hovering over objects of the sliced model in different modes (for example, in the mode of displaying speed or line width). This would greatly simplify the work of preparing the G-code for printing. Below I gave an example of how it might look in the program.

[image]
[image]

supermerill, 2021-12-22:
Problem is, I'm pretty sure there isn't anything to know the first intersection of a ray that is launched from the mouse position.
And if prusa hasn't done it yet, that means it's not easy.

neophyl, 2021-12-22:
Well they requested the same thing on the prusa github too so theres always a chance.