supermerill/SuperSlicer #2106: Add First layer Flow ratio in add settings

liftbag, 2021-12-22:
**Is your feature request related to a problem? Please describe.**
Not exactly a problem, but rather a convenience.
Just as in add settings it is possible to modify the top fill flow ratio, it would be useful, especially during the calibration routines, to be able to vary this parameter on several objects arranged on the bed.

**Describe the solution you'd like**
Just what the title of the feature request indicates

**Additional context**

[image]

supermerill, 2021-12-22:
Sadly, it's not handled the same way as the other ones.
bridges, top, works on the slicing part, on the surfaces.
the first layer is used on the gcode part, for the first layer.

To be able to split it in multiple region... that's the same problem as with speed modifiers. And why the first layer speed also can't be in a modifier.

There is a need of a big revamp of how the gcode writer handles the modifier for this to be implemented.

liftbag, 2021-12-22:
> Sadly, it's not handled the same way as the other ones. bridges, top, works on the slicing part, on the surfaces. the first layer is used on the gcode part, for the first layer.
>
> To be able to split [...]

I understand. So it also explains why there are no retraction speeds in add settings. Those, too, would be very handy when calibrating the retraction.