supermerill/SuperSlicer #2108: RepRap firmware causes long print time

Carib3D, 2021-12-22:
I'm preparing Super Slicer profiles for expected use on a RatRig printer with RepRap firmware . I haven't used it yet because I'm still awaiting the printer kit so I'm just setting up profiles. 

All was going well and I was about finished until I changed the firmware under Print Settings to RepRap. No other changes. After slicing, the quoted print time is WAY longer than Klipper or Marlin. Here are the print time comparisons with everything the same except the choice of firmware:

RepRap 8h9m
Klipper 1h12m
Marlin 1h12m

[SSlicer-RepRap.zip](https://github.com/supermerill/SuperSlicer/files/7764626/SSlicer-RepRap.zip)

I cant say if print time is actually longer since I don't have the printer to use.

supermerill, 2021-12-22:
On reprap firmware, the feedrates are in mm/min. 
So i changed the unit in the gui.
But it was a bad decision, as it creates these kind of problem. I should have just converted the unit at the gcode output.

Carib3D, 2021-12-22:
Oh, I didn't realise that. But how do I correct for it in the slicer? Will this be addressed in a future release? What are the downsides for RepRap firmware if I just leave SS set to Marlin?

supermerill, 2021-12-22:
The impact depends on the choice you have in the top box.
As it impact the time estimation, I guess it's at least that. If you don't correct them (just has to multiply each one by 60), be sure it's on 'disabled'.

Carib3D, 2021-12-23:
Hi, 

 

Many thanks for your fast response. I really hope to resolve this because Super Slicer looks very useful, and I want to use it! 

 

Unfortunately I’m not at a point where it’s practical when set to RepRap firmware, but here’s what I have tried:

 

I wasn’t sure what you meant by ‘top box’ so I tried a few things:

1.	Print Setting/ Speed/ Internal – set to 7200 from 120. Other speeds either 0 or left as % – effect was no change on the estimated print time (8h9m)
2.	Looking at the Gcode preview/ Speed view. The indicated speeds are between 28 and 34 mm/s. Changing to Marlin software, the preview looks identical but the estimated time is down to 1h52m. Switched back to RepRap to try some more things.
3.	Print Setting/ Speed/ Auto Speed. Max print speed changed to 12000 from 200. No change; still 8h9m. Changed Volumetric speed to 1000 from 18. No change so these parameters reverted back.
4.	Print Setting/ Speed/Non-print moves. Change Travel speed xy to 15000 from 250. Bizarrely the print time INCREASED to 10h42m. 

The last one is too strange and made me think something else is going on.

 

What would be the effect if I set the firmware to Marlin even though I run RepRap?

 

 

Regards

Tom Fountain

+1  (246) 231-1377

supermerill, 2021-12-23:
[Light switch housing_reprap_limits_disabled.zip](https://github.com/supermerill/SuperSlicer/files/7769807/Light.switch.housing_reprap_limits_disabled.zip)
[Light switch housing_reprap_limits_fixed.zip](https://github.com/supermerill/SuperSlicer/files/7769808/Light.switch.housing_reprap_limits_fixed.zip)

Carib3D, 2021-12-23:
The issue was resolved by changing the Printer settings/ Machine limits/ Maximum feed rates. These were multiplied by about x60 and then the estimated times came back to a reasonable level.

closed 2021-12-23 by Carib3D