supermerill/SuperSlicer #2111: Unable to input exact value in number sliders

RayPS, 2021-12-23:
### What happened?

On both the `arrangement` & `seam brush size` option sliders, unable to input an exact value, but the same thing does works on PrusaSlicer.
Screen recording:
https://cdn.discordapp.com/attachments/771316909790068756/923481329595736135/Screenshot_2021-12-23_at_15.43.42.mp4

### Project file & How to reproduce

1. Right click on build plate, Add a cube.
2. Right click on `Arrange` Icon.
3. Input some number in the slider.

### Version

2.3.57

### Operating system

MacOS Monterey

### Printer model

Ender-3 V2

supermerill, 2021-12-23:
can't replicate on windows. Will try on linux...

RayPS, 2021-12-23:
According to the discussion in the SS Discord, Windows & Linux doesn't seem to had this issue.

liftbag, 2021-12-23:
I can't replicate it with the latest nightly for macos intel

https://user-images.githubusercontent.com/14921654/147270906-2c1c3daa-9ee1-4fd4-b536-dfbb86469c92.mp4

RayPS, 2021-12-23:
Thanks for the information, I just downloaded Intel version it works fine, I can confirm the issue is only on Arm version.

samwiseg0, 2021-12-23:
> Thanks for the information, I just downloaded Intel version it works fine, I can confirm the issue is only on Arm version.

I never knew this was a bug since prusaslicer behaved the same way. Given that information if this is indeed a bug it is also present on PrusaSlicer arm builds.

RayPS, 2021-12-23:
@samwiseg0 That's strange... Doesn't happen to me on Prusa ARM build.

https://user-images.githubusercontent.com/1941540/147274781-abe4fff8-58df-4866-aca8-3041f61f7632.mp4

supermerill, 2021-12-23:
Maybe they fixed it in the 2.4?

samwiseg0, 2021-12-23:
> @samwiseg0 That's strange... Doesn't happen to me on Prusa ARM build.
> 
>  Screenshot_2021-12-23_at_15.47.34.mp4

You are correct. It does indeed work with PrusaSlicer. 

@supermerill, No it does work on 2.3. I was mistaken. 

In reference to your original issue. I am unable to replicate it. I installed the arm release from github and it works just fine. Could it be an issue with some other software running on your system?

I tested on `12.1 (21C52)` M1 max and `11.6.1 (20G224)` M1

https://user-images.githubusercontent.com/2241731/147292753-2559369e-0209-4832-9858-8e96829f3a07.mov

RayPS, 2021-12-24:
Ok, try these reproduce steps:
1. Open SuperSlicer
2. Right-click -> Add Shape -> Box
3. Press `⌘ + Delete`
4. Click `New Project`, `Discard`
5. Add a box again
6. Arrange option again

I found this issue only happen with/after step 3.
@samwiseg0

RayPS, 2021-12-24:
Update: Tested the new reproduce steps above on Windows PC and Intel Mac, both exist.

samwiseg0, 2021-12-24:
> Update: Tested the new reproduce steps above on Windows PC and Intel Mac, both exist.

I see the same issue now. Another thing that is curious is that I tired the same steps in PrusaSlicer and it does not allow you to start a new project when the box is removed. It seems SuperSlicer is allowing you to start a new project even when there are no objects on the plate.

neophyl, 2021-12-24:
I vaguely remember an issue on the PS github where they discussed reworking the existing/new project functionality due to some issues for 2.4.

liftbag, 2021-12-24:
Update: tested the new reproduce steps on Monterey intel and still it works. The only difference with the new steps is that the the numbers disappears after ⌘ + left click. It's only a cosmetic issue on my side.

With original reproduce steps

https://user-images.githubusercontent.com/14921654/147331048-4d162c36-c3d3-49a7-b10a-fff880a6a226.mp4

With the actual reproduce steps

https://user-images.githubusercontent.com/14921654/147331102-70346b85-57cb-4b55-b2cf-730f05168029.mp4

SuperSlicer version 2.3.58 changes for the nightly macos 447 running on
[image]