supermerill/SuperSlicer #2112: Odd Travel Moves in SS

djngnear, 2021-12-23:
### Discussed in https://github.com/supermerill/SuperSlicer/discussions/2109

<div type='discussions-op-text'>

<sup>Originally posted by **djngnear** December 22, 2021</sup>
I've noticed from SS 2.3.56 (I think it was) that the nozzle will take slight detours instead of going straight to where it needs to go. These pictures are of the same file sliced in SS and PS. You can see the nozzle does slight moves left or right before ending up at the next extrusion. I know it's not the biggest issue but less travel time would mean less oozing filament. And I just sliced it in the latest 2.3.57.8 and there's no change. Maybe someone knows what's going on here? Thanks.
 
[image]

[image]

</div>

djngnear, 2021-12-23:
https://drive.google.com/file/d/1_ymHdMlFttOvfKIZX8Q79fExRPvM8fAc/view?usp=sharing

supermerill, 2021-12-23:
I just test on the 2.3.58 nightly, and by deactivating the 'decelerate with target acceleration', it return to normal.
So i guess the middle point I compute can be wrong. weird. I'll investigate and fix it.

closed 2022-04-01 by supermerill