supermerill/SuperSlicer #2114: Artifacts with spiral vase on a cube of about 20 mm

liftbag, 2021-12-24:
**Bug description**
It is a strange behavior, as it is highlighted in different ways depending on the size of the cube. However, it appears only with measurements close to 20mm per side. It is not present in PrusaSlicer.
To make sure I don't have wrong settings, I set a Prusa printer, a standard prusament PLA filament and 0.20mm QUALITY Print settings.
The defect consists in a misalignment in one of the walls of the cube at a variable height based on the size of the cube.

**Expected behavior**
Walls with all layers perfectly aligned.

!!>> Project File <<!!
[Shape-Box.3mf.zip](https://github.com/supermerill/SuperSlicer/files/7773470/Shape-Box.3mf.zip)

**To Reproduce**
Steps to reproduce the behavior:
1. Open the slicer
2. Right click > Add Shape Box or open a 20mm cube file
3. Set Spiral vase to on.
4. Slice and look for misaligned extrusions
5. Eventually change the measures of the cube in a range between 19 and 21mm and observe the misalignments that occur at different heights

**Screenshots**
21mm shape box
[image]

20mm shape box
[image]

19mm shape box
[image]

**Desktop (please complete the following information):**
OS:
[image]

 - Version: "fix filament override & gapfill NaN" macos 448
 - Printer model: any

sajmonSK, 2021-12-27:
It also exhibits this weird volumetric rate decrease on one side, and the build from 20th december did not do this,
[image]

liftbag, 2021-12-27:
> It also exhibits this weird volumetric rate decrease on one side, and the build from 20th december did not do this, [image]

I can confirm.
[image]

liftbag, 2021-12-29:
Fixed both artifacts and volumetric rate in the latest nightly. Thanks merill 👍

batista16, 2022-01-03:
This did not fix all the issues I have with vase mode: 
[image]
[low-poly-pot-designed-by-additae.zip](https://github.com/supermerill/SuperSlicer/files/7803216/low-poly-pot-designed-by-additae.zip)

supermerill, 2022-01-04:
@batista16 that is an artefact of the process. Can't remove it without specialised slicing (it's because it's slopped: the spiral move the position a bit , and so it's not on the object polygon anymore is there is a slope)
I'll modify the spiral start code to allow you to set the seam position, so you can choose the best start position for it, like the back.

batista16, 2022-01-04:
humm.. 
I think the seam position will not help in every scenario.. there's round objects.. 
I thought that the Spiral Vase entire goal was to have no seam, increasing the Z a full layer height for every layer. 
Cura doesn't have this issue (Although, it has many, many more..), just mentioning cura because they must have solved it somehow right? 

And I have other examples that don't look related to seam, because they are horizontal:
![image](https://user-images.githu
[Craquelado 12x10.zip](https://github.com/supermerill/SuperSlicer/files/7810663/Craquelado.12x10.zip)
busercontent.com/6854591/148130347-edfd8892-6086-4364-ad93-9b2effec1486.png)

supermerill, 2022-01-04:
@batista16 I don't see any problem on this one at the first look.

supermerill, 2022-01-04:
found the issue: there is some detrimental geometry hidden inside the object:
[image]

batista16, 2022-01-05:
Ok, Back to the first example: 
[image]
On Cura:
[image]
[lowpolypot.zip](https://github.com/supermerill/SuperSlicer/files/7814370/lowpolypot.zip)

On this case is completely round, so adjusting the seam position will not solve the issue..

supermerill, 2022-01-05:
can't do much more unless I want to redo entirely the thing.
It slices it in the slope but when moving it up, it jumps over the edge, that's why it is shown as misplaced.

closed 2022-04-01 by supermerill