supermerill/SuperSlicer #2115: no support at some points

Lukas1818, 2021-12-24:
### What happened?

The straight surface get not support at one site.
I am not sure if this intended, but I had expected that supperslicer will create support at the full surface.

[image]

[image]

[image]

### Project file & How to reproduce

[base plate.stl.zip](https://github.com/supermerill/SuperSlicer/files/7774103/base.plate.stl.zip)

* enable support on print bed
* slice 

### Version

 2.3.57.8

### Operating system

Manjaro

supermerill, 2021-12-25:
because you didn't uncheck `dont_support_bridges`

Lukas1818, 2022-01-04:
Thanks
 I had though that this area is the same like the other area, witch get support.

closed 2022-01-04 by Lukas1818