supermerill/SuperSlicer #2117: Superslicer does automatic change values

Lukas1818, 2021-12-24:
### What happened?

Superslicer does automatic change some values.
I did not save the changes, because I would like to keep the current settings.

[image]

### Project file & How to reproduce

[SuperSlicer.zip](https://github.com/supermerill/SuperSlicer/files/7774176/SuperSlicer.zip)

* open SuperSlicer
* close SuperSlicer

### Version

2.3.57.8, 2.3.57.7, 2.3.57.6 and some early too

### Operating system

manjaro

supermerill, 2021-12-25:
width and spacing are exclusive.
If one is active, the other have to be inactive.
In the config, only one have to be set and the other has no value. If it's not the case, the slicer will try to get back to a good state. Here by disabling spacing that are now enable in the default configuration

You can resolve that by saving the configuration, or by adding these lines to your print/default.ini
```
extrusion_spacing = 
external_perimeter_extrusion_spacing = 
external_perimeter_extrusion_spacing = 
first_layer_extrusion_spacing = 
infill_extrusion_spacing = 
perimeter_extrusion_spacing = 
solid_infill_extrusion_spacing = 
top_infill_extrusion_spacing = 
```

Lukas1818, 2022-01-04:
Thanks for clarification.

[image]

If I set a `width` value the associated `spacing` value will be automatic adjust.
But how can a `width` value be set to default but the associated `spacing` value  is unequal to the default?

Lukas1818, 2022-01-04:
thanks for the explanation.

closed 2022-01-04 by Lukas1818