supermerill/SuperSlicer #2118: Ability to enable Z-hop only when travelling over ironed surfaces

melvinmicallef, 2021-12-24:
**Is your feature request related to a problem? Please describe.**
Yes, I use the top ironing pattern or ironing post processing a lot. A problem is that when there is a model which has multiple top layers, they are ironed but afterwards, the nozzle scratches over the ironed surfaces and thus, ruining them. A workaround I'm using is to enable z hop for all the print but this adds print time and sometimes reduces accuracy because of Z backlash.

**Describe the solution you'd like**
Adding an option to this dropdown that is labelled something like 'Only when crossing ironed(both infill and post processing) surfaces'.
[image]

**Describe how it would work**
The slices will be able to detect if the print head is travelling over a top surface that has ironing finished and introduce zhop to that travel move

**Describe alternatives you've considered**
An alternative i see is for the nozzle to take a different route rather than z hopping, however, this might not always be possible since an unfinished part might be surrounded by  an ironed surface and would make re routing impossible. I see z hopping as the best solution

**Additional context**
An example is this model, where the red is an ironed top layer, and the blue arrow is the scratch that will occur on that surface due to zhop being off 
[image]

DVSVIDEO, 2022-01-09:
> **Is your feature request related to a problem? Please describe.** Yes, I use the top ironing pattern or ironing post processing a lot. A problem is that when there is a model which has multiple top [...]

Yes, I have experienced this and it is a very good idea if the developers do it.

DVSVIDEO, 2022-01-09:
Yes, I have experienced this and it is a very good idea if the developers do it.

moarkaboom, 2022-01-19:
I would also appreciate this feature. It doesn't even have to be ironed top layers, but just the top most layer in general.

abcurrie, 2022-05-06:
I too have seen this and it would be nice to have a solution.  Ironing does not seem to be affected by selecting "avoid crossing perimeters".  I have not tried z-hopping to fix this as in general I do not use it.

vampyrex13, 2022-05-10:
+1 on this as I prefer to leave z-hop off to prevent stringing, but I have to enable it for any of my prints that have multiple ironed surfaces.