supermerill/SuperSlicer #2121: (FR) Print progress/remaining time placeholder for gcode

WalnutFox, 2021-12-25:
**Is your feature request related to a problem? Please describe.**
Unable to put progress in custom messages for Klipper.

**Describe the solution you'd like**
A placeholder for both progress percentage and remaining time

**Describe how it would work**
using a gcode placeholder such as **[time_remain]** and **[print_progress]** would work just fine

**Additional context**
building a custom macro for klipper that utilizes these for an output string and would like to have the functionality to use it

related to #1693 and 7ac7a0a

neophyl, 2021-12-25:
Can’t kipper parse the existing time remaining gcodes options to work out what to display ?

supermerill, 2021-12-25:
> A placeholder for both progress percentage and remaining time

A new custom gcode field like the ones for color change?

WalnutFox, 2021-12-25:
> > A placeholder for both progress percentage and remaining time
> 
> A new custom gcode field like the ones for color change?

I'm referring to how we can use [total_layer] to get the total layers but I would like one for remaining time similar to the m117 command used in #1693. It's okay not to include them but since we already have the calculations for time remaining on the I figured it would be relatively easy to save it to a variable that can be output into [time_remain] placeholder the same way we do [total_layer] or [layer_num]. I'm not well versed enough in CPP To do it myself

supermerill, 2022-01-03:
It needs a bit more  work, as the information isn't available.
You can only know the remaining time if you already have created & simulate the full gcode.
So you have to post-process a special tag afterward, likely what's done for the m73.

TypQxQ, 2022-02-21:
I second this.
And if possible and already doing the postprocess, maybe a [time_to_layerchange]. It would be time difference between two toolchanges. 
for adding anticipating heating capabilities to toolchangers.

mrx23dot, 2026-10-28:
Is there an existing way to show any remaining time on marlin LCD?
I stream gcode from PC and neither sides show ETA so I'm flying blind.