supermerill/SuperSlicer #2124: SuperSlices Crashes / Closes without any Error Message

xenon2008, 2021-12-25:
### What happened?

Since i updated to 2.3.57.8 Version. my SuperSlicer always crashes and closes it self.
This happens randomly, if i only view the sliced G-Code, or sometimes if a change Settings or click Slice...

In the Versions before i never had this Problems

### Project file & How to reproduce

i tried difference STLs and always the same
could not say, this happens only when i did this, because this happens Randomly

I use SuperSlicer with a Batchfile with the --datadir "Settings" Paramater.
But i do this a long time with other Versions before .8 and never had such an issue

### Version

2.3.57.8

### Operating system

Windows 10 Pro x64

### Printer model

MK3S,sapphire Pro, all other Printers

btthrt, 2021-12-28:
Same thing. 
There should be a way to get a crash log, but dunno how.

supermerill, 2021-12-29:
It's completely random? 
No way to replicate it?
There is not that much changes between .7 and .8, so if it's really random, I can provide you with a few versions to see where the problem lies.

linux-paul, 2021-12-29:
In my case,
 superslicer crashes while slicing with supports since Version .7. .6 slices without any problems, .8 also crashes. e.g. this data [Webster.zip](https://github.com/supermerill/SuperSlicer/files/7790522/Webster.zip)

BenWhite27, 2021-12-30:
I get a crash to desktop when clicking in this buggy text box if the checkbox is disabled.
[image]

btthrt, 2022-01-01:
> It's completely random?
> No way to replicate it?

It crashes when slicing, almost every time. Rarely it does not.

xenon2008, 2022-01-02:
> 
When I just read that from you, I realized that I had also made a cut on the current component in the past few days

supermerill, 2022-01-03:
The crash is because of the `concentric (filled)` pattern, please don't use this one until the next fix release

closed 2022-04-01 by supermerill