supermerill/SuperSlicer #2126: Crash on missing wipe_only_crossing

RealDeuce, 2021-12-25:
### What happened?

When I selected the "Extruder 1" tab in the printer settings, SuperSlicer crashed with the following backtrace:

```
Thread 1 "slic3r_main" received signal SIGSEGV, Segmentation fault.
Address not mapped to object.
0x000000000264f89c in Slic3r::GUI::Field::toggle (this=0x0, en=false) at /home/admin/src/SuperSlicer/src/slic3r/GUI/Field.hpp:126
126	    inline void			toggle(bool en) { en ? enable() : disable(); }
(gdb) 
(gdb) bt
#0  0x000000000264f89c in Slic3r::GUI::Field::toggle (this=0x0, en=false) at /home/admin/src/SuperSlicer/src/slic3r/GUI/Field.hpp:126
#1  0x00000000026892fc in Slic3r::GUI::TabPrinter::toggle_options (this=0x81547a000) at /home/admin/src/SuperSlicer/src/slic3r/GUI/Tab.cpp:3036
#2  0x0000000002687504 in Slic3r::GUI::Tab::activate_selected_page(std::__1::function<void ()>) (this=0x81547a000, throw_if_canceled=...)
    at /home/admin/src/SuperSlicer/src/slic3r/GUI/Tab.cpp:3605
#3  0x0000000002687270 in Slic3r::GUI::TabPrinter::activate_selected_page(std::__1::function<void ()>) (this=0x81547a000, throw_if_canceled=...)
    at /home/admin/src/SuperSlicer/src/slic3r/GUI/Tab.cpp:2912
#4  0x000000000268fc8d in Slic3r::GUI::Tab::tree_sel_change_delayed (this=0x81547a000) at /home/admin/src/SuperSlicer/src/slic3r/GUI/Tab.cpp:3655
#5  0x000000000269bcd8 in Slic3r::GUI::Tab::create_preset_tab()::$_7::operator()(wxTreeEvent&) const (this=0x8124259e8)
    at /home/admin/src/SuperSlicer/src/slic3r/GUI/Tab.cpp:343
#6  0x000000000269bb6d in wxEventFunctorFunctor<wxEventTypeTag<wxTreeEvent>, Slic3r::GUI::Tab::create_preset_tab()::$_7>::operator()(wxEvtHandler*, wxEvent&)
    (this=0x8124259e0, event=...) at /usr/local/include/wx-3.1/wx/event.h:551
#7  0x000000080833946c in wxEvtHandler::SearchDynamicEventTable(wxEvent&) () from /usr/local/lib/libwx_baseu-3.1.so.5
#8  0x000000080833916b in wxEvtHandler::ProcessEventLocally(wxEvent&) () from /usr/local/lib/libwx_baseu-3.1.so.5
#9  0x0000000808339020 in wxEvtHandler::ProcessEvent(wxEvent&) () from /usr/local/lib/libwx_baseu-3.1.so.5
#10 0x0000000808a8b5a2 in wxScrollHelperEvtHandler::ProcessEvent(wxEvent&) () from /usr/local/lib/libwx_gtk3u_core-3.1.so.5
#11 0x0000000808a9fa61 in wxGenericTreeCtrl::DoSelectItem(wxTreeItemId const&, bool, bool) () from /usr/local/lib/libwx_gtk3u_core-3.1.so.5
#12 0x0000000808a9918d in wxGenericTreeCtrl::OnMouse(wxMouseEvent&) () from /usr/local/lib/libwx_gtk3u_core-3.1.so.5
#13 0x00000008083376af in wxEventHashTable::HandleEvent(wxEvent&, wxEvtHandler*) () from /usr/local/lib/libwx_baseu-3.1.so.5
#14 0x000000080833918d in wxEvtHandler::ProcessEventLocally(wxEvent&) () from /usr/local/lib/libwx_baseu-3.1.so.5
#15 0x0000000808339020 in wxEvtHandler::ProcessEvent(wxEvent&) () from /usr/local/lib/libwx_baseu-3.1.so.5
#16 0x0000000808a8b5a2 in wxScrollHelperEvtHandler::ProcessEvent(wxEvent&) () from /usr/local/lib/libwx_gtk3u_core-3.1.so.5
#17 0x000000080833970c in wxEvtHandler::SafelyProcessEvent(wxEvent&) () from /usr/local/lib/libwx_baseu-3.1.so.5
#18 0x000000080885f9e2 in ?? () from /usr/local/lib/libwx_gtk3u_core-3.1.so.5
#19 0x0000000809807316 in ?? () from /usr/local/lib/libgtk-3.so.0
#20 0x0000000809d6c1f4 in g_closure_invoke () from /usr/local/lib/libgobject-2.0.so.0
#21 0x0000000809d819d9 in ?? () from /usr/local/lib/libgobject-2.0.so.0
#22 0x0000000809d82aaf in g_signal_emit_valist () from /usr/local/lib/libgobject-2.0.so.0
#23 0x0000000809d82f26 in g_signal_emit () from /usr/local/lib/libgobject-2.0.so.0
#24 0x00000008097b3ec9 in ?? () from /usr/local/lib/libgtk-3.so.0
#25 0x000000080964f62f in gtk_propagate_event () from /usr/local/lib/libgtk-3.so.0
#26 0x000000080964f1bf in gtk_main_do_event () from /usr/local/lib/libgtk-3.so.0
#27 0x00000008098ba231 in ?? () from /usr/local/lib/libgdk-3.so.0
#28 0x00000008098efae7 in ?? () from /usr/local/lib/libgdk-3.so.0
#29 0x0000000809e771ce in g_main_context_dispatch () from /usr/local/lib/libglib-2.0.so.0
#30 0x0000000809e77574 in ?? () from /usr/local/lib/libglib-2.0.so.0
#31 0x0000000809e778ca in g_main_loop_run () from /usr/local/lib/libglib-2.0.so.0
#32 0x000000080964eaeb in gtk_main () from /usr/local/lib/libgtk-3.so.0
#33 0x00000008088412b5 in wxGUIEventLoop::DoRun() () from /usr/local/lib/libwx_gtk3u_core-3.1.so.5
#34 0x0000000808206ab2 in wxEventLoopBase::Run() () from /usr/local/lib/libwx_baseu-3.1.so.5
#35 0x00000008081bc326 in wxAppConsoleBase::MainLoop() () from /usr/local/lib/libwx_baseu-3.1.so.5
#36 0x00000008082558a8 in wxEntry(int&, wchar_t**) () from /usr/local/lib/libwx_baseu-3.1.so.5
#37 0x00000000020c7e2b in Slic3r::GUI::GUI_Run (params=...) at /home/admin/src/SuperSlicer/src/slic3r/GUI/GUI_Init.cpp:54
#38 0x00000000011d66fa in Slic3r::CLI::run (this=0x7fffffffe680, argc=1, argv=0x7fffffffe7e8) at /home/admin/src/SuperSlicer/src/PrusaSlicer.cpp:594
#39 0x00000000011d7e09 in main (argc=1, argv=0x7fffffffe7e8) at /home/admin/src/SuperSlicer/src/PrusaSlicer.cpp:798
```

### Project file & How to reproduce

Upgrade to latest, go to printer settings, got to Extruder 1 settings.

The following patch prevents the crash:
```
> git diff
diff --git a/src/slic3r/GUI/Tab.cpp b/src/slic3r/GUI/Tab.cpp
index b290a0fa5..79a512749 100644
--- a/src/slic3r/GUI/Tab.cpp
+++ b/src/slic3r/GUI/Tab.cpp
@@ -3032,7 +3032,7 @@ void TabPrinter::toggle_options()
         }
 
         // wipe_only_crossing can only work if avoid_crossing_perimeters
-        if (!full_print_config.opt_bool("avoid_crossing_perimeters"))
+        if (!full_print_config.opt_bool("avoid_crossing_perimeters") && get_field("wipe_only_crossing", i) != nullptr)
             get_field("wipe_only_crossing", i)->toggle(false);
 
         if (use_firmware_retraction && wipe) {
```

### Version

2.3.57.8

### Operating system

12.3-RELEASE

### Printer model

Voron 2.4

supermerill, 2022-01-04:
Can't reproduce.
Can you zip your configuration directory?
I'm wary to add a check on only one field over the hundreds that are in the method.

RealDeuce, 2022-01-04:
Yeah, I assume there's some missing code that sets a default if it's not present or something along those lines, which is why this isn't a PR.

A quick grep suggests it is/was an issue with saving:
```> grep -r wipe_only_crossing .
./filament/MG Chemicals ABS (Red).ini:filament_wipe_only_crossing = nil
./filament/MG Chemicals ABS (White).ini:filament_wipe_only_crossing = nil
./filament/Reprapper ABS (Black).ini:filament_wipe_only_crossing = nil
./filament/Reprapper ABS (Grey).ini:filament_wipe_only_crossing = nil
./filament/Reprapper ABS (Natural).ini:filament_wipe_only_crossing = nil
./filament/V6 Reprapper ABS (Black).ini:filament_wipe_only_crossing = nil
./filament/BuMat ABS (Brown).ini:filament_wipe_only_crossing = nil
./filament/MG Chemicals ABS (Thermochromic Green).ini:filament_wipe_only_crossing = nil
./filament/HobbyKing ABS (Transparent).ini:filament_wipe_only_crossing = nil
./printer/Voron v2 350 0.4 nozzle.ini:wipe_only_crossing = 1
```

RealDeuce, 2022-01-04:
I've attached some of my config (printer and filament directories), I don't have time to dig through and redact api keys and such.
[partial-config.zip](https://github.com/supermerill/SuperSlicer/files/7809779/partial-config.zip)

RealDeuce, 2022-01-04:
Going through my backups, it seems that it previously wasn't present... December 23rd had no instances:
```
> grep -ir wipe_only_crossing .
>
```

Dec. 24th had just one:
```
> grep -ir wipe_only_crossing .
./filament/Reprapper ABS (Black).ini:filament_wipe_only_crossing = nil
```

Dec 27th, more got added:
```
> grep -ir wipe_only_crossing .
./filament/MG Chemicals ABS (Red).ini:filament_wipe_only_crossing = nil
./filament/MG Chemicals ABS (White).ini:filament_wipe_only_crossing = nil
./filament/Reprapper ABS (Black).ini:filament_wipe_only_crossing = nil
./filament/Reprapper ABS (Natural).ini:filament_wipe_only_crossing = nil
```

And Jan. 1st is when it showed up in my printer config:
```
> grep -ir wipe_only_crossing .
./filament/MG Chemicals ABS (Red).ini:filament_wipe_only_crossing = nil
./filament/MG Chemicals ABS (White).ini:filament_wipe_only_crossing = nil
./filament/Reprapper ABS (Black).ini:filament_wipe_only_crossing = nil
./filament/Reprapper ABS (Natural).ini:filament_wipe_only_crossing = nil
./filament/V6 Reprapper ABS (Black).ini:filament_wipe_only_crossing = nil
./filament/MG Chemicals ABS (Thermochromic Green).ini:filament_wipe_only_crossing = nil
./filament/HobbyKing ABS (Transparent).ini:filament_wipe_only_crossing = nil
./printer/Voron v2 350 0.4 nozzle.ini:wipe_only_crossing = 1
```

supermerill, 2022-01-04:
Even if not in the file, it should be in the slicer from the default value.
Just a note, do not put `compatible_prints_condition = printer_model=~/.*Voron.*/`
printer_model is a printer value, not a print one. It will fail gracefully, only emitting a log but maybe you wanted to write something else?

you can't just zip the `help->show configuration folder`? if you're afraid of sensitive data, you can send it to me on discord. I'll erase it after the fix.

RealDeuce, 2022-01-04:
> Just a note, do not put compatible_prints_condition = printer_model=~/.*Voron.*/
printer_model is a printer value, not a print one. It will fail gracefully, only emitting a log but maybe you wanted to write something else?

Yeah, I was hoping to use the name somehow, but never figured it out... basically, I just fumbled around until things did what I wanted, then I never cleaned it all back up. :smile: 

> if you're afraid of sensitive data, you can send it to me on discord.

Done.

supermerill, 2022-01-30:
merged your patch

commit 23398167 (2022-01-31): Check the return value of get_field() before use

Other calls to toggle() check if the field exists first,
this one should too.

Fixes #2126
commit 0d5add84 (2022-02-01): Check the return value of get_field() before use

Other calls to toggle() check if the field exists first,
this one should too.

Fixes #2126
closed 2022-02-03 by supermerill