supermerill/SuperSlicer #2132: Location of objects doesn't match STL origin

jaggzh, 2021-12-27:
### What happened?

Working from a github build.
I have two objects, aligned in blender, with the same origin point.
They are exported as separate .stl's.

When they're pulled into SS the xy location says they're the same, but they're not.
(Note: It does adjust the Z to lay one on the bed, but the XY should reflect the offset that's applied.)

Here it is in Blender:
[image]

And here they are in SS (xy are displayed as the same for both objects):
[image]

### Project file & How to reproduce

[separate-stl-location-offsets.zip](https://github.com/supermerill/SuperSlicer/files/7781231/separate-stl-location-offsets.zip)

1. Load the two .stl's
1. Compare XY "Positions" 

### Version

github e56be3c

### Operating system

Debian stable

neophyl, 2021-12-27:
Stl files don’t have an origin. The file format by its very nature doesn’t store any external frame of reference. It places the xy bounding box in the centre when imported. Z of course gets placed to bed. 
if you want them to be placed relevant to each other they need to be exported as a single object. Once imported in you can use split to parts or objects if needed to separate them but keep the relative positions in xy.

supermerill, 2021-12-29:
To import multiple objects linked to each other, use "add part" from the contextual menu of the first one".
You can also import all of them in one step and select "single object with multiple parts"