supermerill/SuperSlicer #2134: Gap fill gives nan for extrusion amount in certain cases

luckyphill, 2021-12-28:
### What happened?

Some one on the voron discord (not me, not a github user) reported their print failed when their printer was unable to parse a command.
On inspection, it failed due to the gcode having a nan value for the extrusion amount in a gap fill operation.

This occurred in v2.3.57.8, but did not occur in v2.3.57.5

### Project file & How to reproduce

[Idler_Block.zip](https://github.com/supermerill/SuperSlicer/files/7782123/Idler_Block.zip)

### Version

2.3.57.8

### Operating system

macOS Mojave

### Printer model

voron 2.4

neophyl, 2021-12-28:
Duplicate of https://github.com/supermerill/SuperSlicer/issues/2102 I think.

luckyphill, 2021-12-29:
I think you're right. I'll close it as duplicate

closed 2021-12-29 by luckyphill