supermerill/SuperSlicer #2137: Default Extrusion width grayed out in SuperSlicer 2.3.57.8

zenturacp, 2021-12-29:
**Bug description**
In 2.3.57.8 you cant modify Default extrusion Width, works in .7, i have tried both default profiles and Prusa default profiles

**Expected behavior**
I expect that i can update and set the default extrusion width

**Screenshots**
SuperSlicer 2.3.57.7 (Right)
[image]

SuperSlicer 2.3.57.8 (Wrong)
[image]

**Desktop (please complete the following information):**
 - OS: Windows
 - Version: 2.3.57.8
 - Printer model: All

**Additional context**
I read the release notes and i saw something mentioned about "Updated ConfigManipulation (disabled/enabled) for width & spacing." really not sure what that references to - there is no issue number?

neophyl, 2021-12-29:
It’s this, by design,  default extrusion width value is actually not used except for one specific use case. There was an issues reported where it wasn’t behaving as the user expected so the gui was changed so that it is only editable in that use case. 

im on my tablet at the minute but I’ll try to dig up the issue when I get back to my pc.

supermerill, 2021-12-29:
You can only edit it when there is at least a '0' width field, as this is when it's used.

zenturacp, 2021-12-29:
Ahh super makes sense - but it's not very clear for - I have had 2 people asking me if there is something wrong - maybe some kind of info should be there not to confuse people? It makes sense as you say here - but maybe a hover explained that at last one value should be 0.. Or?

supermerill, 2022-01-04:
I'll modify the tooltip

KapJI, 2022-02-06:
@supermerill please don't forget to update the tooltip 🙂

supermerill, 2022-02-06:
erf

carlbeech, 2023-01-23:
Hi

Sorry - little puzzled over this one... the default extrusion is grayed - if this is the case, where are you supposed to enter the calculated value from Calibration -> Filament Flow Calibration?

Sorry if its a dumb question..

Cheers

Carl

carlbeech, 2023-01-23:
Hi
Sorry, bit more delving - spotted this is actually going into the extrusion multiplier...  so please disregard...

One related query though... I've got two machines - I'm guessing that the multiplier will be different for the two machines - this appears that I'll need to create two PLA profiles... ?   (one for each machine)

Cheers
Carl.