supermerill/SuperSlicer #2138: Height range modifier speed with vase mode - invalid data

jvjvjv78, 2021-12-29:
### What happened?

Error when using height modifier with vase mode

[image] 
[image]
[flow_cube.zip](https://github.com/supermerill/SuperSlicer/files/7788225/flow_cube.zip)

### Project file & How to reproduce

Here is what i did :
1) import stl
2) enable vase mode
3) right clic on object in 3D view, then height range modifier
4) left clic on gear to add speed modification option (external perimeter speed)

Here, if I let default option (which is a speed of 50%), slice can be done without issue, but if I set a speed without the "%",  i got error "The spiral Vase option can only be used when printing single material objects" and "invalid data" instead of export sliced gcode.

When not using vase mode, I can specifiy a speed in mm/s without issue

### Version

2.3.57.5, 2.3.57.8 (I only tried both)

### Operating system

Windows 10

### Printer model

flow_cube.zip attached file

Nicodemous17, 2021-12-30:
I'm trying to do the same thing. I'm getting the same "invalid data" issue as you.

supermerill, 2022-01-04:
duplicate of #190