supermerill/SuperSlicer #2139: Different printtime between Klipper and RepRap on MacOS

jovo0911, 2021-12-29:
### What happened?

There was an issue with printtime. Is different on gcode flavor
printtime for "Standard 20mm calibratin cube" gcode flavor "RepRap Firmware/Sprinter/Lerdge" 1h35m
printtime for "Standard 20mm calibratin cube" gcode flavor "Klipper/Marlin/" 19m

[image]

time for RepRap flavor 1h35m

[image]

time for Klipper flavor 19m

### Project file & How to reproduce

[20mm_calibration_cube_RepRap.3mf.zip](https://github.com/supermerill/SuperSlicer/files/7788563/20mm_calibration_cube_RepRap.3mf.zip)
RepRap time

[20mm_calibration_cube_Klipper.3mf.zip](https://github.com/supermerill/SuperSlicer/files/7788559/20mm_calibration_cube_Klipper.3mf.zip)
Klipper time

### Version

Version 2.3.58.0

### Operating system

macos 13

### Printer model

Hypercube with Afterburner

neophyl, 2021-12-29:
I think this is the same as https://github.com/supermerill/SuperSlicer/issues/2108

jovo0911, 2021-12-30:
I'm going back to 2.3.57.3 show the correct printtime.
There are other problem - "Save Project" use the name from object before, printquality was bad.

supermerill, 2022-01-04:
I took you reprap project and multiplied all limit speed by 60 and i have your klipper time.
Have to make the slicer convert the values when switching from firmware to firmware.