supermerill/SuperSlicer #2142: Bridge flow not used for perimeters in master [e56be3c97]

RealDeuce, 2021-12-29:
### What happened?

Perimeters are not being converted to bridges anymore
[image]
[image]

This results in bad sagging and various other issues.  Setting "No perimeters on bridge areas" to "Remove perimeters" appears to fix this, but it doesn't seem to happen in releases.

### Project file & How to reproduce

[perimeterbridge.zip](https://github.com/supermerill/SuperSlicer/files/7790634/perimeterbridge.zip)

Layers 126 and 127 show it the clearest, but it's also visible in other places.

### Version

2.3.57.8 @ e56be3c97

### Operating system

FreeBSD 12.2-RELEASE-p3

### Printer model

Voron 2.4 with AB-BN 30

RealDeuce, 2021-12-29:
Layer 153 has it even worse, even with "Remove perimeters", the bridge is not detected, and it's printed as perimeters.

So it looks like this is really an issue with it being binned as an overhang, not a bridge.  If I set overhangs_width_speed to non-zero and overhangs_width, it uses overhang bridge flow and seems to print OK, but it never actually detects those features as a bridge.

supermerill, 2022-01-04:
Your overhang detection is deactivated
[image]
enter something greater than 0, if you want a  very small value, choose something like 15%.

supermerill, 2022-01-07:
if you want only bridge,s and no perimeter, you can give the 'no perimeters on bridge areas' -> 'remove perimeters' a try

RealDeuce, 2022-01-07:
> if you want only bridge,s and no perimeter, you can give the 'no perimeters on bridge areas' -> 'remove perimeters' a try

Even with that, on layer 153, the feature is not detected as a bridge, only an overhang, so is not changed with that setting:
[image] (Overhangs disabled)

Using overhang detection, it's detected as an overhang, but it still doesn't get bridge attachment:

[image]

supermerill, 2022-01-07:
you know, an overhang is a bridged perimeter.
for the second layer, because of the "only one perimeter on top" so it want solid fill and not perimeters. But some of it is over the void, so some of it is transformed into bridge.

RealDeuce, 2022-01-08:
> you know, an overhang is a bridged perimeter.

I don't really understand this statement... while all bridges involve an overhang, not all overhangs involve a bridge... or I wouldn't expect extra_perimeters_overhangs or support_material_threshold to work unless at least overhangs_width_speed was applied.

My understanding of a bridge vs. an overhang is that a bridge has a segment with 100% unsupported width that has a lower percentage of unsupported width on both sides of that segment, while an overhang is anything with more than 0% unsupported width.

"Keep only bridges" does seem to work better, but it doesn't seem to go quite far enough... it would be nice if there was an option to attach unattached bridges as well.  In the attached file for example, layer 106 of stealthburner_toolhead_(dragon)-front-beta0.stl, 
[StealthBunerAndCW2.zip](https://github.com/supermerill/SuperSlicer/files/7832748/StealthBunerAndCW2.zip)

[image]

If those two bottom U-shaped bridge extrusions (that start and end in mid-air) were extended until they reached either a perimeter or infill (or, removed), the part would print better.

But maybe I'm conflating a few different issues.

RealDeuce, 2022-02-01:
Ok, I think I understand this all now more or less, thanks!

closed 2022-02-01 by RealDeuce