supermerill/SuperSlicer #2150: Feature Request: Add cost of kWh into Cost estimation

DgitalTrex, 2021-12-30:
### Version
2.3.57.8

### Operating system type + version
Windows 10 (build 19044), 64-bit edition

### Behavior
Sliced Info > Cost, should take into account cost of kWh usage based on estimated printing time. kWh should have a per printer profile value. When cost per kWh is not set on printer profile, cost of kWh is ignored.

_Is this a new feature request?_ YES

Juusoz, 2021-12-30:
I doubt that would be easy to implement. The kWh spent is based off of your printer's bed size, how hot it is, how hot the hotend is, how much power the motors are using, how much power the screen is using, any lights, fans and even the loss due to an inefficient power supply.

If you want an accurate reading on your kWh used, you should just buy a power meter for your socket. they cost like $15 and give you way more accurate readings. Get the average from that and multiply by the estimated hours and you'll get a pretty good guess.

DgitalTrex, 2022-01-03:
No I have a power meter so I can "guestimate" my kWh usage. I want to enter
than "guesstimate" in the software itself so it guesstimate for me

troypouliot, 2022-01-03:
I agree this would be helpful for us that run businesses to provide more accurate quoting.

DgitalTrex, 2022-01-03:
I'm thinking it would be hidden in the advanced printer profile. default 0 (disables it)

Then the cost would add calculations somewhere like
```
if kwhcost > 0:
    cost += time*costkwh
    display costkwh (right below the total filament usage)
    display cost
```

supermerill, 2022-01-04:
but won't that depend mainly on the bed temp?
so printing pla vs abs is vastly different.

neophyl, 2022-01-04:
Agreed it would need to be stored in the filament profiles and not in the print or printer ones as its dependent on the printing temps and has nothing to do with the print or printer settings.

I can see that if you add this then it would also be best to plan in adding more placeholders for filenames etc at the same time, as I'm sure once its available they will want them :)  

I mean if you have the time and you know the kwh then its basic math.  This is basically for those who cant be bothered to do the math themselves.

supermerill, 2022-01-04:
the problem is that you need multiple variables to have a good enough estimation:
* kwh per second for printer use without heatbed
* ambient temp
* kwh per second for the bed at ambient temp + 40
then I can infer a rough estimate.

troypouliot, 2022-01-04:
I don't think it needs to be that complicated. It should be the users responsibility to measure and input the kwh _while actively printing_ then the software just needs to multiply that by the print time. I don't think we need to worry about ambient temp or what the printer is consuming while idle or warming up. Of course it wont be exact, but it would be much closer than nothing.

I have my printer profiles divided up by material type so having it as a printer setting would work _for me_.

troypouliot, 2022-01-04:
"Machine $/hour" would work for me. Maybe the OP has some other requirements and use cases though.

DgitalTrex, 2022-01-04:
> So it's mostly "print time" * "fixed factor" ? Why using the kwh then, you can directly put your "$ per hour", so you can integrate the maintenance, depreciation expense, etc.

$/hr works. I just want it specifically tagged for electricity just in case some other feature is requested in the future. Such as machine depreciation/hr, rent/hr, etc.

I do agree that it should be filament based. Considering vastly different temps needed for diff filaments. User will have to be responsible in getting this value though.

BentHaase, 2022-10-16:
How is the current state, where can this option be found? I must have overseen it somewhere :blush:

BentHaase, 2022-10-16:
> How is the current state, where can this option be found? I must have overseen it somewhere blush

Yup, found it. :facepalm: 

[image]

closed 2024-02-11 by supermerill