supermerill/SuperSlicer #2155: Show extruder change colors in thumbnails

grantemsley, 2021-12-31:
**Is your feature request related to a problem? Please describe.**
I have a dual extruder printer, and often print single objects in 2 colors by changing extruders at a set height. But that change isn't reflected in the thumbnails. The thumbnails always show the color of the main extruder.

Separate objects *do* show the color change in the thumbnails, but not single objects with extruder changes.

**Describe the solution you'd like**
Show the color changes in the thumbnails.

I've uploaded an example project in case I'm not clear. In the thumbnails it shows completely purple, but it should be half purple and half white.

[Shape-Box.zip](https://github.com/supermerill/SuperSlicer/files/7795136/Shape-Box.zip)

gandy92, 2022-10-18:
I can confirm this - when multiple extruders are defined, colours of filaments assigned to each extruder are shown correctly in the thumbnail (i.e. it works correctly when a tool-change is involved). However, if a colour change is configured for a specific extruder at a specific height, only the first colour is used in the thumbnail for all heights.

gandy92, 2022-10-18:
From what I can see from the sources, thumbnails are rendered by ``GLCanvas3D::_render_thumbnail_internal()`` in ``slic3r/GUI/GLCanvas3D.cpp``, where each volume is assigned with a "uniform_color" according to the configuration options. 

However, I'm not quite able to figure out if height-dependent colour changes split a volume in sub-volumes, which should make it possible to assign individual colours to each sub-volume.