supermerill/SuperSlicer #2156: Wipe cannot be switched off

SphaeroX, 2021-12-31:
### What happened?

I have tried turning off the wipe function, but it makes no difference whether it is selected or not.

No retract is executed either.

### Project file & How to reproduce

Look at this Video: https://youtu.be/qwoYtXZcqfk

And also the ProjectFile

[wipe.zip](https://github.com/supermerill/SuperSlicer/files/7795656/wipe.zip)

### Version

2.3.57.8

### Operating system

windows 11

### Printer model

custom

SphaeroX, 2021-12-31:
[image]

 There is the result

supermerill, 2022-01-04:
The issue (it's very difficult to pinpoint) is that currently the wipe only happen when there is a retraction.
You have the "only retract when crossing perimeter" checked in the printsetting -> infill section.
So you have almost no wipe. Uncheck this option to see the real wipe happening.
[image]
Here you can see that the yellow short lines show wipe before each travel now.
For the external perimeter, you can see it go back inside before, and for 2 reason:
 - hide the seam
 - prevent wiping on the external perimeter.
That's why when you disable the wipe, there is still the 'seam reduction' move.
Note that with 'extra wipe', these things are move a bit father away.

Now, you complain that this move is detrimental to your seam. 
Can you edit your gcode file to remove that move at the end of the external perimeter extrusion. It's easy, just remove the line between 'WIPE_START' and 'WIPE_END'  lines after 'External perimeter' (need verbose setting):
```
G1 X119.792 Y108.619 E2.7252 ; External perimeter
G1 X117.907 Y108.618 E2.78368 ; External perimeter
;WIPE_START
G1 X117.447 Y108.618 F12000 ; move inwards before travel
;WIPE_END
G92 E0 ; reset extrusion distance
```
then print with the new file. Is the result greatly improved? If yes, I'll add a setting to remove that.

SphaeroX, 2022-01-05:
Thank you for your good explanation, I was just wondering why I can't completely deactivate the wipe.

I have now done the test, once with and once without "only retract when crossing perimeter". In the third test I also commented out the wipe completely.

By the way: If anyone wants to test this, here is the RegEx to replace it: ;WIPE_START\n.*\n.*\n.*\n;WIPE_END

It looks like it's more of a printer problem, the result never changed.

[image]

supermerill, 2022-01-05:
thank you for doing the tests!

Kesiunas, 2022-01-20:
> Here you can see that the yellow short lines show wipe before each travel now. For the external perimeter, you can see it go back inside before, and for 2 reason:
>
> * hide the seam
> * prevent wiping [...]

Hello,

Sorry if this will be off the topic but how have you managed to get wiping before every travel move (particularly on the inside perimeters)? Because I've managed to set retractions before every move but option wipe while retracting is on and looking at the gcode preview I see no wipe lines even if I turn them on to show in the menu.

I've attached config if this would be easier to check: https://we.tl/t-bNp8ZG9Ceu 

[image]

Thank you very much:)

supermerill, 2022-01-20:
@Kesiunas 
Hi
Currently, this wipe is done while retracting, so if your retraction is extremely small, there is an extremely small wipe. You can make it longer with a very slow retraction.
Also, if you set to use 100% of the retraction before the wipe, there is no retraction time left to wipe, so no wipe.

Modify your config:
0% `retract_before_wipe`
uncheck `wipe_only_crossing`
`retract_length` to 0.1mm
`retract_speed` to 1mm/s

Kesiunas, 2022-01-20:
Thank you very much for quick response and your support for this slicer! I really appreciate it.

Next time I will open new discussion.
Thank you