supermerill/SuperSlicer #2157: Portions missing from sliced object

n9jcv, 2021-12-31:
When slicing this object
https://www.thingiverse.com/thing:909901
stl as seen in Sketchup
[image]

sliced object as seen is SS 2.3.57
[image]

There are portions of the object that are missing.  There are four towers.  All the same height.  SS renders 2 of them full height and 2 of them are shorter.

 When the above object is loaded SS, it reports dimensions. Then I slice in SS, .28 layers and .4 nozzle, I get the last layer at 19.88, 71st layer.  Only 2 towers go to 19.88, 2 of the towers stop at 17.64.  In Sketchup, looking at the stl, the model appears to be identical for all 4 towers.

It seems perhaps, that the fine detail is too much for PS or my .4mm nozzle. 

This is probably not a bug, but due to the fact I would need a smaller nozzle? But how can SS render 2 towers to top and not all 4? SS does label the tops of the 2 complete towers as thin wall, so this may have something to do with it. I do have print thin walls enabled with default settings.

So, I then changed nozzle size to .2mm and layer height to .1, and still just 2 towers rendered fully. Resolution is set to zero and internal resolution is .2.

Just fyi, Prusa 2.3.0, stops rendering all 4 towers at 17.64mm.

Windows 10
Voxelab Aquila - Ender 3 clone

supermerill, 2022-01-04:
can't reproduce
[image]

can you post your project file?

n9jcv, 2022-01-07:
Sure, I guess I should have posted the 3mf originally sorry.
[TOWERBUG.zip](https://github.com/supermerill/SuperSlicer/files/7831413/TOWERBUG.zip)

supermerill, 2022-01-08:
still can't reproduce.
Are you sure it's the right project?
what version of 2.3.57 did you used?

n9jcv, 2022-01-08:
I downloaded the zip above   extracted the 3mf   Open SS 2.3.57.7 
then click slice
same issue as above   2 towers render full height and 2 do not

full towers in red   short towers in blue circles   
on Windows 10

I would be open to a web screen share/zoom/something if needed    or I can answer more questions    
[image]
[image]

n9jcv, 2022-01-08:
as a further comment your pic shows all 4 towers rendered short.  just fyi

neophyl, 2022-01-21:
With 2.3.57.7 I get the same as the OP, 2 towers rendered with thin wall extrusion and 2 without that are shorter.  Rotating the part makes no difference.

I also tried in 2.3.57.9.  That slices the part with all 4 towers short, no thin wall extrusions so matches Merill's pics.  Looks like there is a difference between .7 and .9 but that's not too surprising as the software has been changed after all.