supermerill/SuperSlicer #2159: Hi SS,

GAMERSW0RLD, 2022-01-01:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

**Describe how it would work**
Be as descriptive as your skill level allows you.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or paint drawing about the feature request here.

GAMERSW0RLD, 2022-01-01:
PS. Happy New Year :)

supermerill, 2022-01-04:
You can slow down the top fill. Should do what you want.

If you really want to slow down 'the last layer', then 
* via a right-clic add a 'height range modifer'
* set the start at height at the height of the last layer and the stop at height to the height of the last layer + the layer height
* add speed modifiers to it to change all the speed you want to slow down. (should be easier in next release, with a master speed)