supermerill/SuperSlicer #2162: MacOs arm broken

BlazingLink, 2022-01-01:
### What happened?

When I open the 2.3.57.8 MacOs arm version, Mac Os tells me that "Superslicer is broken and cannot be opened" The non-arm version opens fine
[image]

### Project file & How to reproduce

just download and open superslicer 2.3.57.8 MacOs arm version

### Version

2.3.57.8

### Operating system

MacOs Monterey

Johnno88, 2022-01-02:
I too am having the same issue with MacOs ARM Version

Operating System
MacOS BigSur

MacBook Pro 13" M1 2020

viljamip, 2022-01-02:
Same here 
[image]
Macbook Pro 16" M1 Max 2021
MacOs Monterey

Interesting thing is that when I then tried to download and install 2.3.57.7 which used to work, I got the same error. Maybe this can be something related to some MacOs update?

viljamip, 2022-01-02:
Ah, I'm silly. I forgot this step:
to allow it to run : xattr -cr /Applications/SuperSlicer.app

I managed to get the latest build running, but it appears to be 2.3.57.7.
Version: 2.3.57.7
Build: SuperSlicer_2.3.57.7_2021-12-22

Johnno88, 2022-01-03:
So,,,  your replies are not clear, 

Will MacOS Arm 2.3.57.8 be fixed or do we run the command line after we install it?

viljamip, 2022-01-03:
For me this happened:
I downloaded the MacOs ARM build from the 2.3.57.8 release. I dragged it to my Applications folder and then I ran `xattr -cr /Applications/SuperSlicer.app`. When I launched the app, it works fine but it reports its version to be 2.3.57.7. Not sure if the version number is wrong or if that is the same build as before.

supermerill, 2022-01-03:
It's the version numbering that is wrong. I forgot to build it with the new one. It wasn't added to the auto build process at the time and I borked it.

bapski, 2022-01-09:
so, decided to try out Superslicer on my macbook pro m1 and i too am having this issue on SuperSlicer_2.3.57.9_macos_arm_220105

how do i get superslicer running?

viljamip, 2022-01-09:
> so, decided to try out Superslicer on my macbook pro m1 and i too am having this issue on SuperSlicer_2.3.57.9_macos_arm_220105
> 
> how do i get superslicer running?

Open Terminal and run `xattr -cr /Applications/SuperSlicer.app`. Then you can launch the app normally.

BlazingLink, 2022-01-20:
> Ah, I'm silly. I forgot this step: to allow it to run : xattr -cr /Applications/SuperSlicer.app

Same Lol

closed 2022-01-20 by BlazingLink