supermerill/SuperSlicer #2171: tool change not happening outside ooze shield

robtrama, 2022-01-04:
### What happened?

when ooze prevention is enabled the tool change doesn't happen outside the ooze shield when hotend offsets are defined in firmware and causes some oozing to track along with with heat creep to occur as it sits inside the ooze shield once several layers happen. causes a hotbox type of scenario where heat from the bed and hotends build up 

### Project file & How to reproduce

[dual extrusion tests.zip](https://github.com/supermerill/SuperSlicer/files/7805152/dual.extrusion.tests.zip)

### Version

2.3.57.7

### Operating system

windows 10

### Printer model

Robo R2 with chimera hotend

supermerill, 2022-01-04:
You need a "cross the draft shield on toolchange" setting?

robtrama, 2022-01-04:
something like that would be useful, where you can input the amount it
crosses the draft shield by (extruder offset essentially)

supermerill, 2022-01-04:
If you set your tool change gcode to something like
```
G1 X0 Y0
T[next_extruder]
```
does it resolve your problem (I know it's too far away, but it's just to understand exactly what's needed).

robtrama, 2022-01-04:
i will test this out as soon as i can and let you know

robtrama, 2022-01-05:
hello,

it does solve the problem, i had to make it X22 Y0 as it wanted to crash
when it switched to the second extruder (its offset is +20mm in x
direction) and I had to add a G92 as it was not adding it in. I attached a
zip with 3 gcode outputs and have them labeled. first tool change starts at
line 2043
looks like this

G1 X22 Y0
T[next_extruder]
G92 E0 ; reset extrusion distance

supermerill, 2022-01-05:
ok.
A "move extruder outside of skirt for toolchange" in the print-mm tab should do the trick

robtrama, 2022-01-05:
That would be awesome,  thank you for all the quick responses 

Robert Trama
Fodera Guitars Partners
***@***.***
646-734-3655