supermerill/SuperSlicer #2172: Calibration gone crazy

manjaroblack, 2022-01-04:
### Version
2.3.57.8

### Operating system type + version
Linux Mint 20.2

### Behavior
SuperSlicer is progressively going crazy. I tried to go through the calibrations yesterday and succeeded in performing the bed leveling and filament flow calibrations. When I tried to do the filament temperature calibration, klipper kept giving me an error saying "unable to parse move: G1 X- Y5 F5900". I searched the gcode and confirmed that this pattern is common all through out the generated code. Today I attempted to perform the bed leveling again after changing my nozzle. The generated preview looked insane and all over the place. 

[image]

_Is this a new feature request?_
No

#### Project File (.3MF) where problem occurs
[First layer calibration.3mf.zip](https://github.com/supermerill/SuperSlicer/files/7805257/First.layer.calibration.3mf.zip)

supermerill, 2022-01-04:
gcode precision -> xyz decimals, set it to something resonable like the default value of 3.

I think ther eis still a problem with 0. Maye the absence of the '.' bugs out the formatter. Will put a min of 1 on it.

supermerill, 2022-01-04:
I fixed the formatter instead.
But you shouldn't set it to 0 unless you have a house-printing printer.

manjaroblack, 2022-01-05:
Wow, yeah that was the issue. I am not sure how that got changed. I must have deleted the default by accident!

closed 2022-01-05 by manjaroblack