supermerill/SuperSlicer #2176: Line width overhang perimeter

gogsi1, 2022-01-05:
### Version
2.3.57.8 

### Operating system type + version
Windows 10 Pro 21H2

### Behavior
Perimeter: 0.4
external Perimeter: 0.4
[image]

Width does not increase

Perimeter: 0.6
external Perimeter: 0.6
[image]
[image]

neophyl, 2022-01-05:
Please attach your zipped up project file as requested when creating a new issue.

supermerill, 2022-01-05:
That's because the bridge width doesn't change from perimeter width but from bridge flow ratio.
But I don't advise you to try to print bridge wider than your nozzle diameter, that will create extreme sagging.
You can remove the perimeters to have only bridges with 'no_perimeter_unsupported_algo' -> 'no perimeters'
you can also use a height range modifier and modify the perimeter width for this layer only.