supermerill/SuperSlicer #2178: pass stls to already running SuperSlicer instance

Lukas1818, 2022-01-05:
**Is your feature request related to a problem? Please describe.**
Every time you open a stl a new SuperSlicer instance will be created. If you download multiple stls for slicing from internet (as example Thingiverse or Oneshape) and chose "open with" at you browser, a new SupesSlicer window will appear. Now the additional windows must be closed manual and the stls must be add to the running SuperSlicer manual to slice all at once.

**Describe the solution you'd like**
A option to pass stls to the already running Superslicer.

**Describe how it would work**
Be as descriptive as your skill level allows you.

supermerill, 2022-01-05:
Did you try the preference settign "allow just a single instance of superslicer" ?

Lukas1818, 2022-01-05:
Thanks 
I had check the settings before creating this issue, but I have overlook this.

closed 2022-01-05 by Lukas1818