supermerill/SuperSlicer #2180: No shadow in thumbnails when black is used

msdossajo, 2022-01-05:
### What happened?

When generating the thumbnails and the filament color is black no "shades/shadows" are generated and it's hard to identify the object

### Project file & How to reproduce

Printer Settings > General >Color override disabled
Filament Settings > Filament > Color set o black

[image]
[image]

### Version

2.3.57.9

### Operating system

Linux Mint 19.3 Cinnamon

### Printer model

Ender 3 V2 running Klipper (v0.10.0-220-g608cd38d) Fluidd (v1.16.2)

neophyl, 2022-01-05:
Erm, don’t use black then ?  It’s hard to do darker shadows when the colour is already the darkest there is.

supermerill, 2022-01-06:
I guess I can make "dark grey" the darkest colour possible, so there is still a bit of leeway for shadowing...