supermerill/SuperSlicer #2182: FreeCAD support does not work with AppImage FreeCAD

VorpalBlade, 2022-01-05:
### What happened?

After setting the path to FreeCAD (AppImage version for Linux x86-64) I expect the FreeCAD support the Generate menu to work.

However it just reports the following error:
```
Error, cannot find the freecad (version 0.19 or higher) python at '/home/arvid/bin/freecad/bin/python', please update your freecad python path in the preferences.
```

From what I can tell, SuperSlicer expects to be able to locate the python interpreter used by freecad based on the path to the program (somehow). The AppImage version of freecad contains the python binary embedded inside it and it is mounted as a fuse file system under /tmp when running FreeCAD.

For example
```console
$ freecad                
/tmp/.mount_freecaLT5oYJ
FreeCAD 0.19, Libs: 2021.1015R24301 +4280 (Git)
Failed to establish dbus connection
^Z
$ bg
$ ls /tmp/.mount_freecaLT5oYJ/usr/bin/python
/tmp/.mount_freecaLT5oYJ/usr/bin/python
```

Another approach is needed to support the AppImage version of FreeCAD.

Even without that I don't see how it would work on Linux in general: if using a distro installed version of FreeCAD, /usr/bin/python from the distro is used, but at least on Arch Linux, the freecad modules are not installed into the standard python path (instead they are installed under /usr/lib/freecad/Mod). I haven't tested this case though.

### Project file & How to reproduce

[empty_project.3mf.zip](https://github.com/supermerill/SuperSlicer/files/7817626/empty_project.3mf.zip)

### Version

2.3.57

### Operating system

Arch Linux (SuperSlicer installed using AUR package)

### Printer model

Prusa Mk3s

supermerill, 2022-01-05:
I only tested it on windows.
I need the path to the freecad python executable, so I can run it. I don't know how to access it when it's an appimage.
and it assumes that it can access the `../../lib` (freecad lib) from it.

Do you have any clue on input on how to access that into an appimage?

VorpalBlade, 2022-01-07:
Hm, that might be tricky!

### The AppImage case
My understanding is that AppImages are basically a small wrapper executable + a compressed file system image. The wrapper mounts the file system image, then executes the main binary. The file system image is unmounted when the program exits.

I will look into if it is possible to manually mount/unmount the AppImage. I presume it must be. Then you will I presume need to find the proper python path under that. Using the python console inside FreeCAD gave me this monstrosity:
```python
>>> import sys
>>> sys.path
['/tmp/.mount_FreeCaOnZU6N/usr/Mod/asm3', '/tmp/.mount_FreeCaOnZU6N/usr/lib/python3.9/site-packages/git/ext/gitdb', '/tmp/.mount_FreeCaOnZU6N/usr/Mod/appimage_updater', '/tmp/.mount_FreeCaOnZU6N/usr/Mod/Web', '/tmp/.mount_FreeCaOnZU6N/usr/Mod/Tux', '/tmp/.mount_FreeCaOnZU6N/usr/Mod/Test', '/tmp/.mount_FreeCaOnZU6N/usr/Mod/TechDraw', '/tmp/.mount_FreeCaOnZU6N/usr/Mod/Surface', '/tmp/.mount_FreeCaOnZU6N/usr/Mod/Start', '/tmp/.mount_FreeCaOnZU6N/usr/Mod/Spreadsheet', '/tmp/.mount_FreeCaOnZU6N/usr/Mod/Sketcher', '/tmp/.mount_FreeCaOnZU6N/usr/Mod/Show', '/tmp/.mount_FreeCaOnZU6N/usr/Mod/Robot', '/tmp/.mount_FreeCaOnZU6N/usr/Mod/ReverseEngineering', '/tmp/.mount_FreeCaOnZU6N/usr/Mod/Raytracing', '/tmp/.mount_FreeCaOnZU6N/usr/Mod/Points', '/tmp/.mount_FreeCaOnZU6N/usr/Mod/Path', '/tmp/.mount_FreeCaOnZU6N/usr/Mod/PartDesign', '/tmp/.mount_FreeCaOnZU6N/usr/Mod/Part', '/tmp/.mount_FreeCaOnZU6N/usr/Mod/OpenSCAD', '/tmp/.mount_FreeCaOnZU6N/usr/Mod/MeshPart', '/tmp/.mount_FreeCaOnZU6N/usr/Mod/Mesh', '/tmp/.mount_FreeCaOnZU6N/usr/Mod/Measure', '/tmp/.mount_FreeCaOnZU6N/usr/Mod/Material', '/tmp/.mount_FreeCaOnZU6N/usr/Mod/Inspection', '/tmp/.mount_FreeCaOnZU6N/usr/Mod/Import', '/tmp/.mount_FreeCaOnZU6N/usr/Mod/Image', '/tmp/.mount_FreeCaOnZU6N/usr/Mod/Idf', '/tmp/.mount_FreeCaOnZU6N/usr/Mod/Fem', '/tmp/.mount_FreeCaOnZU6N/usr/Mod/Drawing', '/tmp/.mount_FreeCaOnZU6N/usr/Mod/Draft', '/tmp/.mount_FreeCaOnZU6N/usr/Mod/Arch', '/tmp/.mount_FreeCaOnZU6N/usr/Mod/AddonManager', '/tmp/.mount_FreeCaOnZU6N/usr/Mod', '/tmp/.mount_FreeCaOnZU6N/usr/lib', '/tmp/.mount_FreeCaOnZU6N/usr/lib64', '/tmp/.mount_FreeCaOnZU6N/usr/Ext', '/tmp/.mount_FreeCaOnZU6N/usr/bin', '/tmp/.mount_FreeCaOnZU6N/usr/lib/python39.zip', '/tmp/.mount_FreeCaOnZU6N/usr/lib/python3.9', '/tmp/.mount_FreeCaOnZU6N/usr/lib/python3.9/lib-dynload', '/tmp/.mount_FreeCaOnZU6N/usr/lib/python3.9/site-packages', '/home/arvid/.FreeCAD/Macro', '/tmp/.mount_FreeCaOnZU6N/usr/Macro', '/tmp/.mount_FreeCaOnZU6N/usr/lib/python3.9/site-packages/gitdb/ext/smmap']
```

### The distro binary case
Here the issue is basically that you will need to discover the proper `PYTHONPATH` env variable to use, since the system Python executable is used. For Arch Linux at least it looks like everything is installed under /usr/lib/freecad (no idea if that is the same on other distros, lets hope so). Presumably you will need to set `PYTHONPATH=/usr/lib/freecad/Mod/Arch:/usr/lib/freecad/Mod/Draft:...`.

How do you deal with finding the proper python path currently? Or does it work completely differently on Windows?

psiberfunk, 2022-01-13:
This sure sounds like an edge case does it not ?  Why not just install freeCAD outside an app image ... App Images aren't particularly friendly in many respects to being called externally for precisely the reasons listed here.

VorpalBlade, 2022-01-13:
The reason for me in particular to use AppImage is that I use an experimental nightly build of FreeCAD. This is newer than what the official Arch Linux repositories carry. Building FreeCAD from source takes an inordinate amount of time (it is a huge project), so I want to avoid that. That leaves me with the option of downloading the official nightly AppImage.

Another reason to not use the distro package is if you are on a slower moving distro (for example Ubuntu LTS (Long Term Support)) that only updates to new releases every other year.

supermerill, 2022-01-13:
maybe you can extract the content in a dir in your home?
like https://superuser.com/questions/1301583/how-can-i-extract-files-from-an-appimage
Then you can still run it and now other app can access it.
Bonus: you can install plugins inside it now!

VorpalBlade, 2022-01-19:
Hm. I haven't tried it yet, but I already see some issues with that:

1. The python path relative to FreeCAD would still be different than you expect from Windows from what I can tell. (You seem to expect `<path to freecad from settings>/bin/freecad` when it is  `<path to freecad from settings>/usr/bin/freecad`.
2. FreeCAD plugins work fine anyway, they are installed into `~/.FreeCAD` (I guess that would be `c:\Users\<user>\AppData\Roaming\FreeCAD` or something like that on Windows?)
3. What about FreeCAD installed via the distro package manager (which is what most users probably use)?
4. FreeCAD is huge on Linux. The AppImage is 880 MB, the extracted version is 3 GB! (Not too bad of a compression ratio though!)

Also it is not really a seamless process, especially when I use the nightly builds of FreeCAD (as I would have to update that extracted version).

If you still believe it is worth pursuing the extracted version I can give it a try though.

supermerill, 2022-01-19:
I can work on path discovery, if something is sub-optimal or problematic.
To be able to use the appimage, Maybe I can extract it in a temp dir and then delete it after use?

If you want something small and light, It's maybe better to work on something similar for openscad.

VorpalBlade, 2022-01-23:
> I can work on path discovery, if something is sub-optimal or problematic. To be able to use the appimage, Maybe I can extract it in a temp dir and then delete it after use?

It seems that executing the AppImage using `--appimage-mount` would give you what you want there, with the added benefit of not having to wait for it to actually extract it (nor taking up any disk space for it). There is also `--appimage-extract` if for some reason you actually want to extract it.

> If you want something small and light, It's maybe better to work on something similar for openscad.

The upstream OpenSCAD distributes it as an AppImage on Linux too. Of course there are also the distro provided packages (which generally installs it as /usr/bin/openscad). However I believe this is a much simpler case indeed, as you always want the main program here (openscad), as opposed to a possibly bundled side program (python).