supermerill/SuperSlicer #2183: Perimiter acceleration is used for travel moves in some cases.

bithoarder, 2022-01-05:
### What happened?

Travel moves appear to be using perimeter acceleration when moving between separate objects, it looks like the send half of a travel move is using different acceleration that the first half.
The attached project have a extremely low perimeter acceleration to emphasize  the problem.

### Project file & How to reproduce

[TravelWithPerimiterAcceleration.zip](https://github.com/supermerill/SuperSlicer/files/7817867/TravelWithPerimiterAcceleration.zip)

### Version

2.3.57.9

### Operating system

Ubuntu 20.04

### Printer model

Voron v0.1

CCS86, 2022-01-07:
Without looking at your code, it sounds like you are describing a new feature to avoid ringing in perimeters by "arriving" with the perimeter accel value. If you show up to the external perimeter at full travel decel, it will cause ringing.

supermerill, 2022-01-07:
can you try the nightly like this one? https://github.com/supermerill/SuperSlicer/actions/runs/1651975415
in the acceleration section, you can disable it.

closed 2022-04-01 by supermerill