supermerill/SuperSlicer #2186: First layer width doesn't adjusted based on first layer height

erictioh, 2022-01-06:
### What happened?

First layer width is currently intact despite i adjust the first layer height. When i adjust first layer height higher and the first layer width remain the same, it causes issue where it cannot stick well on bed because width is not enough.

### Project file & How to reproduce

see the what happen

[image]

### Version

2.3.57.9

### Operating system

Windows 10

### Printer model

Creality CR10 V3

neophyl, 2022-01-06:
Not sure I'm understanding the issue.  Those settings are independent in Super Slicer.  In Prusa Slicer they are linked but not with SuSi.  If you want your first layer extrusion width to be wider then just set it to the width you want.

supermerill, 2022-01-06:
Also, they removed the link in PS 2.4.

Which tooltip makes you think the width is dependant on the height? I have to fix it.

erictioh, 2022-01-06:
Ok, you can see that the tool tip for first layer stated that this is influenced by default layer height. And i did a test indeed that if i increase the default layer height this first layer width is increased automatically.
[image]

Then very intuitively i will ask myself, it should be that first layer width influenced by first layer height isn't it ? that's what lead me to open this ticket.

neophyl, 2022-01-06:
Ah I see.  There's the relationship between width and spacing.  Changing one changes the other in the settings fields.  The exact relationship though does take into account the layer height as well as the overlap percentage.  So its not a direct change, its changing the math between the 2.
So for example a first layer width of 150% at a layer height of 0.1 produces a spacing of 145%, increase the layer height to 0.2 and the same width of 150% produces a spacing of 139%

Not sure what wording would make that better though.

erictioh, 2022-01-06:
Yes check the screenshot of the 2 value of layer height which does change the first layer spacing or first layer width (depends on what i update either at the width or spacing before hand).

[image]
[image]

however my question is since layer height does change width/spacing, why not first layer height change the first layer width and spacing ?

erictioh, 2022-01-06:
layer height do change almost all the param in the Extrusion width, however first layer height doesn't change a bit of any width, the impact is the higher first layer heigh is introduced, i have to manually guessing calculate the first layer width. However this is not the case for the rest of layer where the slicer calculate that out automatically for me. 

Not sure the way i see is correct or perhaps something hidden that i don't aware, you guys can further comment

neophyl, 2022-01-06:
I'm not sure I understand your question so I'll leave that to Merill.

supermerill, 2022-01-07:
that's because you set the first layer width and not the spacing. SO the width is the field 'set' and the spacing is the field 'computed from width and layer height'
the other fields are the opposite: spacing is set and width is computed.
You can look at the colour of the label to see who's set (light grey means computed)
You can have the meaning of the colour (and change it) when clicking on the '?'