supermerill/SuperSlicer #2190: unable to change "Extrustion Width: Default"

Fam-of3, 2022-01-06:
### What happened?

Recently updated to the new releases 2.3.57.8 as well as 2.3.57.9. I can'not change, adjust a few settings anymore in the Print Tab. My profile i had have all greyed out default boxes. didn't have problems with version 2.3.57.7 ; am able to load the old version and make an adjustment, then re-enter the latest versions and have the settings change however still grey out.

### Project file & How to reproduce

empty new file

### Version

2.3.57.9

### Operating system

windows 10

### Printer model

Creality cr-10v2

neophyl, 2022-01-06:
Do you mean this ? https://github.com/supermerill/SuperSlicer/issues/2137

Fam-of3, 2022-01-06:
Yes, thank you for ticket find

closed 2022-01-06 by Fam-of3