supermerill/SuperSlicer #2191: Odd perimeters generated underneath embossed layer

nchantal, 2022-01-07:
### Version
2.3.57.8

### Operating system type + version
Windows 10 Home

### Behavior
I'm trying to print a simple label that has a rectangular base with embossed text. On the layer prior to the text, the slicer generates these internal perimeters that make the printed label look a lot less clean because they stick out from underneath the embossed text.
[image]

I tried this with similar models where the embossed part is simpler. The slicer also generates some internal perimeters and fill for these, but the upper layers cover these features up.
[image]

#### Project File (.3MF) where problem occurs
[FranceLabelTest.zip](https://github.com/supermerill/SuperSlicer/files/7825380/FranceLabelTest.zip)

neophyl, 2022-01-07:
Turn off single perimeter on top layers tickbox.  Its more complicated than that, theres some other settings you can tweak and there are a couple of other posts around here somewhere with the same issue (so might be closed).  I dont have the time to search right now.  Turning off the option though should clear them right up though.

neophyl, 2022-01-07:
Just found this in the discussion section , someone else having the same problem https://github.com/supermerill/SuperSlicer/discussions/1896  
New thread too.

supermerill, 2022-01-07:
you can also just set the "only one periemter" -> 'min width' to 0

nchantal, 2022-01-07:
Thank you both, that did it!

closed 2022-01-07 by nchantal