supermerill/SuperSlicer #2192: Spiral vase - incomplete layers

krankydonkey, 2022-01-07:
### What happened?

When using spiral vase mode, the software creates large gaps in certain layers, resulting in an easily breakable part.

[image]

### Project file & How to reproduce

[Spiral Vase Error.zip](https://github.com/supermerill/SuperSlicer/files/7825588/Spiral.Vase.Error.zip)

### Version

2.3.57.9

### Operating system

Windows 11

### Printer model

Voron 2.4

supermerill, 2022-01-07:
uncheck 'gap_fill_last'

I'll add it to the list of things that need to be switch down for vase mode

krankydonkey, 2022-01-08:
ah, that'll do it! Cheers

closed 2022-04-01 by supermerill