supermerill/SuperSlicer #2194: Machine Limits Dialog show incorred units

thetekguy1, 2022-01-07:
### Version
2.3.57

### Operating system type + version
Manjaro Linux, latest rolling build (up to date) running Gnome 41.2

### Behavior
Printer-Machine Limits dialog under Maximum and Minimum Feedrates are all listed as mm/min, should be listed as mm/s

supermerill, 2022-01-08:
why?

thetekguy1, 2022-01-08:
they should be corrected to state mm/s, otherwise people will input feedrates into those fields in mm/minute which is a 60x larger value, throwing off print estimates wildly. 40 mm/s = 2400 mm/min...

supermerill, 2022-01-08:
https://github.com/supermerill/SuperSlicer/issues/440

thetekguy1, 2022-01-08:
this issue is in the 2.3.57.9 release, if 440 was fixed and closed in 2020,
the issue is back...

thetekguy1, 2022-01-08:
yes, this is  repeat of #440 from 2020, it has resurfaced in 2022 in the current 2.3.57.9 release.

thetekguy1, 2022-01-09:
no

thetekguy1, 2022-01-09:
what do you mean by no?

closed 2022-01-20 by thetekguy1