supermerill/SuperSlicer #2195: Bridge Infill apearing on top layer

DaLeberkasPepi, 2022-01-08:
### What happened?

If the standard width & flow setting is set at 0 there appear bridge infills on the top layers of my model.
[image]

Changing that setting to 100% or 0.4 or going back to the version 2.3.57.3 fixes this for me.

Here is the .3mf file i got that problem with:
[stealthburner_toolhead_(rapido)-rear-cw2-beta1.zip](https://github.com/supermerill/SuperSlicer/files/7831629/stealthburner_toolhead_.rapido.-rear-cw2-beta1.zip)

### Project file & How to reproduce

Under Print Settings > Width & Flow > Set Standard to "0"
If you change it to 100% or a 0.4 for example the bridge infill on top layer doesnt appear anymore.

### Version

tested with 2.3.57.7-9

### Operating system

Windows 11

### Printer model

Voron 2.4

supermerill, 2022-01-08:
it'(s too thin and there is a pixel unsupported.
you can use a modifier to add a perimeter here or remove the bridge_infill_margin ;  see fixed version here.
[2195_stealthburner_toolhead_(rapido)-rear-cw2-beta1_fixed.zip](https://github.com/supermerill/SuperSlicer/files/7832152/2195_stealthburner_toolhead_.rapido.-rear-cw2-beta1_fixed.zip)
.

DaLeberkasPepi, 2022-01-08:
mhm but this same file does work fine in 2.3.57.3 (without all the workarounds) thats why i was thinking that this could be a bug.

supermerill, 2022-01-26:
I don't have the time for that, but I guess that the 2.3.57.3 has a bug.